raspberrypi/rpicam-apps #379: Allow specifying non-color/greyscale modes

kralo, 2022-10-04:
My camera sensor is BW only, 

```
Available cameras
-----------------
0 : ov9281 [1280x800] (/base/soc/i2c0mux/i2c@1/ov9281@60)
    Modes: 'R8' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                  1280x720 [144.40 fps - (0, 40)/1280x720 crop]
                  1280x800 [120.76 fps - (0, 0)/1280x800 crop]
           'R10_CSI2P' : 640x400 [260.42 fps - (0, 0)/1280x800 crop]
                         1280x720 [144.40 fps - (0, 40)/1280x720 crop]
                         1280x800 [120.76 fps - (0, 0)/1280x800 crop]
```

how can I force it to be used in 8bit mode?

`libcamera-vid -n --mode 1280:800:10:P -v` does not even work (even if it's the auto-selected format), libcamera apparently tries color formats, i.e.
```
[pid  1757] ioctl(14, VIDIOC_TRY_FMT, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, fmt.pix={width=1280, height=800, pixelformat=v4l2_fourcc('N', 'V', '1', '2') /* V4L2_PIX_FMT_NV12 */, field=V4L2_FIELD_NONE, bytesperline=0, sizeimage=0, colorspace=V4L2_COLORSPACE_SMPTE170M}} => {fmt.pix={width=1280, height=800, pixelformat=v4l2_fourcc('N', 'V', '1', '2') /* V4L2_PIX_FMT_NV12 */, field=V4L2_FIELD_NONE, bytesperline=1280, sizeimage=1536000, colorspace=V4L2_COLORSPACE_SMPTE170M}}) = 0
[2:04:39.751928648] [1757] ERROR Camera camera.cpp:1023 Can't configure camera with invalid configuration

```

Would you consider adding GREY support, maybe here:

https://github.com/raspberrypi/libcamera-apps/blob/22a52590c33a813743b4e6337478c208201c77b1/core/libcamera_app.cpp#L42-L61

naushir, 2022-10-04:
Yes, this is a problem we have noticed as well.  I think the solution would be a change to the libcamera pipeline handler rather than the code you pointed at above - effectively accounting for mono sensor formats in the mode selection.  I'll work on this shortly.

naushir, 2022-10-19:
With the latest release of libcamera/libcamera-apps in apt, this will now be fixed.

However... if you were to update, you do need to do one extra thing to make the ov9281 work:
```
sudo cp /usr/share/libcamera/ipa/raspberrypi/ov9281.json /usr/share/libcamera/ipa/raspberrypi/ov9281_mono.json
``
I've posted a change to the libcamera dev list with this change as well, but until it gets merged, you will have to do this manually.

kralo, 2022-10-20:
Thank you. I tried with  ` libcamera-jpeg -n -v --mode 1280:800:8:U -o test3.jpg` , looks correct.

closed 2022-10-20 by kralo