raspberrypi/rpicam-apps #381: [BUG] ERROR message: *** no cameras available *** after libcamera update

supergreditur, 2022-10-04:
**Describe the bug**
After updating the libcamera library. to the latest version i get the following error when running libcamera-hello:

```
Made DRM preview window
[0:11:35.780766133] [745]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
ERROR: *** no cameras available ***
```

On previous versions the camera is detected and works as expected.

I am currently using a raspberry Pi 3B+ with the [5MP OV5647 Noir Camera](https://www.arducam.com/product/b003301-arducam-5mp-ov5647-1080p-noir-camera-for-raspberry-pi-infrared-camera-module-sensitive-to-ir-light/).

**Bug report**
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 04-10-2022 (13:51:19)
Command: /usr/bin/camera-bug-report -t 5 -o camera-bug-report -c libcamera-hello

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
r-and-d-develop2

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000aee0dd1b
Model		: Raspberry Pi 3 Model B Plus Rev 1.3

--------------------------------------------------------------------------------
[... 829 lines cut ...]
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Aug 26 2022 14:04:36 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

** libcamera-hello --version **
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:21:46)
libcamera build: v0.0.0+3866-0c55e522

--------------------------------------------------------------------------------
libcamera-hello
--------------------------------------------------------------------------------
Made DRM preview window
[0:07:22.215233955] [691]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
ERROR: *** no cameras available ***

```

**Additional context**
Add any other context about the problem here.

naushir, 2022-10-04:
You seem to be running a Debian release rather than a Raspberry Pi OS release:
```
** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
You ought to get in touch with their support channels to see what may have caused this breakage.

6by9, 2022-10-04:
Memory says the 64bit version of Raspberry Pi OS does report as being Debian.
Noting that this is a 64bit OS is significant though.

naushir, 2022-10-04:
Ah, apologies, I did not realise that!

supergreditur, 2022-10-04:
It is indeed the 64bit lite version of Raspberry Pi OS that we use.

To be more precise, this is the working (old) version of Raspberry Pi OS lite (64-bit): https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/

And this is the new version of Raspberry Pi OS lite (64-bit) where the camera no longer seems to work: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-26/

Note that updating libcamera on the 'old' OS version also seems to breaks the application.

naushir, 2022-10-05:
I've just tried with a fresh 64-bit os image with all updates and everything seems to work as expected:

```
pi@pi4:~ $ libcamera-hello --version
libcamera-apps build: 1bf0ccacf2ef 31-08-2022 (16:21:49)
libcamera build: v0.0.0+3866-0c55e522
pi@pi4:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
```

As expected our libcamera/libcamera-apps versions are identical.  Would you be able to try with another sensor and/or a different camera connector cable?

supergreditur, 2022-10-06:
I've tried 3 different raspberry Pi 3B+ and 2 different cameras (the [5MP OV5647 Noir Camera](https://www.arducam.com/product/b003301-arducam-5mp-ov5647-1080p-noir-camera-for-raspberry-pi-infrared-camera-module-sensitive-to-ir-light/) and the [RPIZ-CAM-GS](https://www.berrybase.de/en/Pixelpdfdata/Articlepdf/id/4628/onumber/RPIZ-CAM-GS) ), all resulted in the exact same error.

I tried reinstalling the OS (both the old and the new) about 4 times and each time the result was consistent. Old image works, but the new one fails to recognize the cameras.

However, I also tried a raspberry Pi Zero w2, and that seems to work fine! **It seems like the update only breaks on the Raspberry Pi 3B+**

naushir, 2022-10-06:
> However, I also tried a raspberry Pi Zero w2, and that seems to work fine! **It seems like the update only breaks on the Raspberry Pi 3B+**

Thanks for the update!  Confirmed that a fresh 32/64-bit image does not work for me on the 3B+ either.  Investigating...

naushir, 2022-10-06:
@supergreditur, are you able to do a ``rpi-update`` on your device and try this again?  Note that this will update your kernel and firmware to the very latest versions, so please do make a backup of your card if there is anything important on it.

There has been some recent changes to the camera probe routine on boot that are available.  With those changes, my 64-bit image now seems to work as expected, i.e. ov5647 works fine.  Would you be able to try this and report back if it works for you?

supergreditur, 2022-10-06:
This indeed fixed this issue!
Thanks a lot!

You may mark this issue as resolved