raspberrypi/rpicam-apps #384: [BUG] Libcamera-still/jpeg crash

zahigeni, 2022-10-12:
Hey all,
**Describe the bug**

**we running Libcamera-still from python via subprocess in 10 seconds intervals :**

subprocess.run(["libcamera-still", "-n", "-o", image_location, "--width", "1000", "--height", "754"])

**and after a time period ( its occurs randomly, some times its happening after boot) 
we getting** 

INFO Camera camera manager.CD:293 libcamera v0.0.0+3424-e680fle\nERROR: *** no cameras available ***n'

only reboot solves the issue 
also swap memory is disabled 

**did not had the chance to use the bug-report-tool when this problem occurs this is the output in journalctl:**

Oct 11 14:20:23 he01097 kernel: Unable to handle kernel paging request at virtual address 0000000000001008
Oct 11 14:20:23 he01097 kernel: Mem abort info:
Oct 11 14:20:23 he01097 kernel:   ESR = 0x96000005
Oct 11 14:20:24 he01097 kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Oct 11 14:20:24 he01097 kernel:   SET = 0, FnV = 0
Oct 11 14:20:24 he01097 kernel:   EA = 0, S1PTW = 0
Oct 11 14:20:24 he01097 kernel: Data abort info:
Oct 11 14:20:24 he01097 kernel:   ISV = 0, ISS = 0x00000005
Oct 11 14:20:24 he01097 kernel:   CM = 0, WnR = 0
Oct 11 14:20:24 he01097 kernel: user pgtable: 4k pages, 39-bit VAs, pgdp=000000001a8c6000
Oct 11 14:20:24 he01097 kernel: [0000000000001008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Oct 11 14:20:24 he01097 kernel: Internal error: Oops: 96000005 [#1] PREEMPT SMP
Oct 11 14:20:24 he01097 kernel: Modules linked in: algif_hash algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc>
Oct 11 14:20:24 he01097 kernel: CPU: 2 PID: 1105 Comm: libcamera-jpeg Tainted: G         C        5.10.92-v8+ #1514
Oct 11 14:20:24 he01097 kernel: Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Oct 11 14:20:24 he01097 kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
Oct 11 14:20:24 he01097 kernel: pc : kernfs_fop_open+0x6c/0x398
Oct 11 14:20:24 he01097 kernel: lr : kernfs_fop_open+0x44/0x398
Oct 11 14:20:24 he01097 kernel: sp : ffffffc012543ab0
Oct 11 14:20:24 he01097 kernel: x29: ffffffc012543ab0 x28: ffffffc012543c88 
Oct 11 14:20:24 he01097 kernel: x27: 0000000000000000 x26: ffffff80063ffcf0 
Oct 11 14:20:24 he01097 kernel: x25: 0000000000000000 x24: ffffff801a82ed00 
Oct 11 14:20:24 he01097 kernel: x23: ffffffc01038c530 x22: 0000000000000000 
Oct 11 14:20:24 he01097 kernel: x21: ffffffc010b28040 x20: ffffff8008f2c000 
Oct 11 14:20:24 he01097 kernel: x19: 0000000000001000 x18: 0000000000000000 
Oct 11 14:20:24 he01097 kernel: x17: 0000000000000000 x16: 0000000000000000 
Oct 11 14:20:24 he01097 kernel: x15: 0000000000000000 x14: 0000000000000000 
Oct 11 14:20:24 he01097 kernel: x13: 0000000000000000 x12: 0000000000000030 
Oct 11 14:20:24 he01097 kernel: x11: 0000000000000030 x10: 0101010101010101 
Oct 11 14:20:24 he01097 kernel: x9 : ffffffc01038c574 x8 : 7f7f7f7f7f7f7f7f 
Oct 11 14:20:24 he01097 kernel: x7 : ffffffc012543c10 x6 : 0000000000000000 
Oct 11 14:20:24 he01097 kernel: x5 : ffffff8008f2c000 x4 : ffffff8008f2c004 
Oct 11 14:20:24 he01097 kernel: x3 : 0000000000000000 x2 : 0000000000000001 
Oct 11 14:20:24 he01097 kernel: x1 : ffffffc01038d2e0 x0 : 0000000000000001 
Oct 11 14:20:24 he01097 kernel: Call trace:
Oct 11 14:20:24 he01097 kernel:  kernfs_fop_open+0x6c/0x398
Oct 11 14:20:24 he01097 kernel:  do_dentry_open+0x134/0x398
Oct 11 14:20:24 he01097 kernel:  vfs_open+0x34/0x40
Oct 11 14:20:24 he01097 kernel:  path_openat+0xa18/0xe28
Oct 11 14:20:24 he01097 kernel:  do_filp_open+0x84/0x100
Oct 11 14:20:24 he01097 kernel:  do_sys_openat2+0x1f8/0x2a8
Oct 11 14:20:24 he01097 kernel:  do_sys_open+0x60/0xa8
Oct 11 14:20:24 he01097 kernel:  __arm64_sys_openat+0x2c/0x38
Oct 11 14:20:24 he01097 kernel:  el0_svc_common.constprop.3+0xc0/0x1f8
Oct 11 14:20:24 he01097 kernel:  do_el0_svc+0x2c/0x98
Oct 11 14:20:24 he01097 kernel:  el0_svc+0x20/0x30
Oct 11 14:20:24 he01097 kernel:  el0_sync_handler+0x90/0xb8
Oct 11 14:20:24 he01097 kernel:  el0_sync+0x180/0x1c0
Oct 11 14:20:24 he01097 kernel: Code: b40015e0 f94026a1 52800020 b4001521 (b9400a61) 
Oct 11 14:20:24 he01097 kernel: ---[ end trace f63e4b96704c7504 ]---
Oct 11 14:20:24 he01097 NetworkManager[1003]: <info>  [1665498024.5908] manager: startup complete
Oct 11 14:20:26 he01097 systemd[1]: systemd-fsckd.service: Succeeded.
Oct 11 14:20:27 he01097 systemd-timesyncd[350]: Initial synchronization to time server 162.159.200.123:123 (2.debian.pool.ntp.org).
[... 1106 lines cut ...]
--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux he01097 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Jan 20 2022 13:58:22 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

** libcamera-hello --version **
libcamera-apps build: 0dc5ea038c79 04-03-2022 (10:35:06)
libcamera build: v0.0.0+3424-e68e0f1e

naushir, 2022-10-12:
Thank you for the report.

First off, your libcamera/libcamera-apps version is very out of date.  I suggest you either run an update (``sudo apt update && sudo apt upgrade``) or start again with the most reason Raspberry Pi OS image release and try this again.

naushir, 2022-10-17:
Are there any updates on this issue?  If not, I will close this down.

naushir, 2022-10-18:
Closing due to inactivity.  Please re-open if you still have problems.

closed 2022-10-18 by naushir

zahigeni, 2022-10-18:
hey @naushir for now I started from the newest bullseye os image and running my app , its happen randomly and I'm waiting to see if the issue repeat itself with the updated os, will update if do

naushir, 2022-10-18:
Thank you!  Do let me know if you run into the problem again.