raspberrypi/rpicam-apps #385: [Feature] Allow control of audio sample rate

pakal, 2022-10-12:
For now, the sample rate (Hz) of captured audio is not editable by command line option, as noted in https://forums.raspberrypi.com/viewtopic.php?t=333940

It'd be useful to have that audio, e.g. to lower the rate for performance/memory considerations  :)

naushir, 2022-10-14:
This was always in the plan - just ran out of time to implement it.

I'll look to pick it up again when I get the time.

naushir, 2022-10-14:
Turns out I had already written pretty much all of the code, expect exposing a command line argument.
https://github.com/raspberrypi/libcamera-apps/pull/390 seems like a plausible change to get this working.  Would you be able to test this out?

pakal, 2022-10-16:
Thanks,

I checked out your branch and built it like noted here : https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps

But these custom  apps don't support any of the libav-related CLI options :

```
pi@raspberrypi:~/licamera_naushir/build $ libcamera-vid --timeout 15000 --nopreview  --output file.mp4 --codec libav  --libav-audio
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
ERROR: *** unrecognised option '--libav-audio' ***
```

I guess i'm missing some dependencies and/or cmake flags, is there somewhere some info on how to build a feature-complete libcamara-apps please?

pakal, 2022-10-17:
Ok so in case it helps anybody, this required installing libavcodec-dev, libavutil-dev, libavformat-dev and libavdevice-dev with apt under Raspberry Pi OS  ^^'

I've tested the new parameter on newly compiled executable :
` ./libcamera-vid --timeout 30000 --nopreview  --output file.mp4 --codec libav  --libav-audio --audio-samplerate 16000`

It seems to work fine  :)

[image]

naushir, 2022-10-18:
Thanks for confirming.  The change will make it into apt very soon.

I'll close this issue down now.

closed 2022-10-18 by naushir