raspberrypi/rpicam-apps #387: [BUG] MP4 container gets corrupted if libcamera-vid capture is interrupted by ctrl-C

pakal, 2022-10-13:
I've seen several CLI options of control pause/resume of capture with signals or keyboard, but not for interactively stopping the recording.

I'm not sure what's supposed to happen in case of quitting before the timeout (or when timeout=0), using ctrl-C.

To me the video recording should stop and the container should be finalized, but when I tested it on raspberry pi 3B+, I got an invalid mp4 file (for both vlc and mediaplayerclassic) instead.

```
pi@raspberrypi:~ $ libcamera-vid -t 0 --nopreview -o output3_truncate.mp4 --codec libav --libav-audio
[0:36:47.592549052] [3545]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:36:47.712083479] [3557]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:36:47.713831447] [3545]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-YUV420
[0:36:47.714878166] [3557]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
[h264_v4l2m2m @ 0xdca770]  <<< v4l2_encode_init: fmt=181/0
[h264_v4l2m2m @ 0xdca770] Using device /dev/video11
[h264_v4l2m2m @ 0xdca770] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0xdca770] requesting formats: output=YU12 capture=H264
[h264_v4l2m2m @ 0xdca770] Failed to set number of B-frames: Invalid argument
[h264_v4l2m2m @ 0xdca770] Failed to set gop size: Invalid argument
Input #0, pulse, from 'default':
  Duration: N/A, start: 1663809092.049613, bitrate: 1536 kb/s
    Stream #0:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
Output #0, mp4, to 'output3_truncate.mp4':
    Stream #0:0: Video: h264 (h264_v4l2m2m), drm_prime(tv, smpte170m/smpte170m/bt709), 640x480, q=-1--1, 200 kb/s, 1000k tbn
    Stream #0:1: Audio: aac (LC), 48000 Hz, stereo, fltp, 32 kb/s
<ctrl-c>
```

naushir, 2022-10-14:
The Ctrl+C signal is not trapped by libcamera apps. This will caused a truncated mp4 container, which as you pointed out will not play correctly.

Instead, you could use the ``--keypress`` options in ``libcamera-vid``, and when you want to stop recording press ``X`` or ``x`` followed by ``Enter``.  That should clean up correctly.  Is that suitable for your usage?

naushir, 2022-10-14:
It's a trivial change to fix ``libcamera-vid`` to catch the ``Ctrl+C`` signal and exit cleanly.  I'll prepare a patch shortly.

```diff
diff --git a/apps/libcamera_vid.cpp b/apps/libcamera_vid.cpp
index 6dd24ada2e86..0eaeb8e0708b 100644
--- a/apps/libcamera_vid.cpp
+++ b/apps/libcamera_vid.cpp
@@ -28,6 +28,8 @@ static void default_signal_handler(int signal_number)
 static int get_key_or_signal(VideoOptions const *options, pollfd p[1])
 {
        int key = 0;
+       if (signal_received == SIGINT)
+               return 'x';
        if (options->keypress)
        {
                poll(p, 1, 0);
@@ -76,6 +78,7 @@ static void event_loop(LibcameraEncoder &app)
        // Monitoring for keypresses and signals.
        signal(SIGUSR1, default_signal_handler);
        signal(SIGUSR2, default_signal_handler);
+       signal(SIGINT, default_signal_handler);
        pollfd p[1] = { { STDIN_FILENO, POLLIN, 0 } };
```

naushir, 2022-10-18:
This has now been merged.

closed 2022-10-18 by naushir