raspberrypi/rpicam-apps #393: [BUG] streaming video using UDP, on the Raspberry Pi (server).

overheat, 2022-10-20:
log on the VLC:
```
$ vlc udp://@:34235 :demux=h264
...
[h264 @ 0x13c9b3800] error while decoding MB 32 6, bytestream -6
[h264 @ 0x13c930e00] error while decoding MB 37 7, bytestream -6
[h264 @ 0x13caa5a00] error while decoding MB 13 7, bytestream -5
[h264 @ 0x13cb42400] error while decoding MB 16 6, bytestream -6
[h264 @ 0x13cabc400] error while decoding MB 17 7, bytestream -6
[h264 @ 0x13caf9800] error while decoding MB 15 7, bytestream -6
[h264 @ 0x13c9b3800] error while decoding MB 0 15, bytestream -6
[h264 @ 0x13c930e00] error while decoding MB 24 6, bytestream -6
[h264 @ 0x13caa5a00] error while decoding MB 20 6, bytestream -6
...
```

log on raspberry Pi zero 2 W:
```
$ libcamera-vid -t 0 --inline -o udp://192.168.31.119:34235 --codec h264
Made DRM preview window
[1:30:40.037162380] [5410]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[1:30:40.107153165] [5411]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[1:30:40.109056654] [5410]  INFO Camera camera.cpp:1035 configuring streams: (0) 640x480-YUV420
[1:30:40.109819049] [5411]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
```
[image]

overheat, 2022-10-21:
Thanks!

```
Run the ffplay or VLC client in another terminal on the server. This eliminates the network and uses the loopback interface.
```
This is the best quality one. But there is a 1~2 second delay on VLC or ffplay compare to the libcamera-vid original view. Cloud you kindly explain more?

naushir, 2022-10-21:
There will always be a small delay due to ffplay buffering the stream.  You can minimise this buffering by using the following command line options with ffplay:
```
ffplay tcp://<ip-addr-of-server>:<port> -fflags nobuffer -flags low_delay -framedrop
```

overheat, 2022-10-25:
> There will always be a small delay due to ffplay buffering the stream. You can minimise this buffering by using the following command line options with ffplay:
> 
> ```
> ffplay tcp://<ip-addr-of-server>:<port> -fflags nobuffer -flags low_delay -framedrop
> ```

yes, it is the minimal latency playback.

closed 2022-10-25 by overheat