raspberrypi/rpicam-apps #395: [HOW-TO] Troubleshoot Intermittent Timeouts

N2ZLC, 2022-10-25:
I'm having intermittent timeout issues capturing images. It happens in both Python and on the CLI. For example, if I execute...

```while :; do libcamera-jpeg --nopreview -o test.jpg; sleep 30; done```

...I get a low number of random successes but always a timeout in the end...

```
pi@test:~ $ while :; do libcamera-jpeg --nopreview -o test.jpg; sleep 30; done
INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received

[SUCCESS]

INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received

[SUCCESS]

INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
INFO Camera camera.cpp:1026 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Still capture image received

[SUCCESS]

INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
WARN V4L2 v4l2_videodevice.cpp:2004 /dev/video0[12:cap]: Dequeue timer of 254314000.00us has expired!
ERROR RPI raspberrypi.cpp:1862 Unicam has timed out!
ERROR RPI raspberrypi.cpp:1863 Please check that your camera sensor connector is attached securely.
ERROR RPI raspberrypi.cpp:1864 Alternatively, try another cable and/or sensor.
ERROR: Device timeout detected, attempting a restart!!!

[FAILURE]
```

I'm not touching the camera or the pi, which are both mounted on a sturdy chassis. The cable that connects them is short—10cm or about 4 inches. I'm using a Raspberry Pi 3 Model A Plus Rev 1.0 headless. The camera is a Raspberry Pi High Quality HQ Camera - 12MP. I'm running the current stable release of Bullseye. It's not the power supply. I've tried several including the official one that puts out 3A.

Any ideas on how I can narrow down the root cause?

naushir, 2022-10-26:
Thanks for the report.  Can I ask a few questions to get a better idea on what may be going on:

1) How long after starting the application does the timeout occur?  Do you receive any frames from the sensor before the timeout?  You can check this by using the ``-v`` command line argument.
2) Once this goes wrong, can you successfully restart the application to get it working, or do you need to do a full reboot or power cycle?
3) Do you happen to have any other cameras that you could use in place of the imx477?
4) What kind of environment are you running in, i.e. could there be some sort of mechanical/electrical/radio interference that causes the timeout?

N2ZLC, 2022-10-26:
I really appreciate you responding, and your questions are prescient—particularly #3.

I swapped out the IMX477 with an IMX519 with all else being equal and had no issues.

I then swapped out the IMX519 with another IMX477 and had no issues.

The problem was indeed the original camera. Sad, because it was brand new out of the box (so far as I know).

What made me reject the idea that it was the camera was that it worked around 66-80% of the time (it would fail after 2 or 3 or 4 images). No reboot would be required to try again.

Thanks again.

closed 2022-10-27 by N2ZLC

naushir, 2022-10-27:
> The problem was indeed the original camera. Sad, because it was brand new out of the box (so far as I know).

Is that an official Raspberry Pi camera purchased through one of our resellers?  If so, you should be able to return it as faulty for a replacement.