raspberrypi/rpicam-apps #400: [BUG] HQ camera not detected after initial boot with a RP3

cpixip, 2022-10-31:
An original HQ camera connected by to a RP3 with freshly installed and updated OS is not detected when booting up from a "power off" state. Rebooting _without disconnecting the power_ does yield a functional camera. 

Circumstances:

- brand new OS install on RP3 (download 30.10.2022 via RP-imager, updated system after first boot)
- Shutdown and switch off power after install
- RP3 connected again to power, boots up
- trying to access the HQ camera yields:

```
pi@raspi-06:~ $ libcamera-hello 
Made X/EGL preview window
[0:00:35.121130298] [1367]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
ERROR: *** no cameras available ***
```

- rebooting without disconnecting the power yields after the reboot:

```
pi@raspi-06:~ $ libcamera-hello 
Made X/EGL preview window
[0:00:39.702628265] [1402]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+21-7c855784
[0:00:39.786416702] [1418]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:00:39.787508421] [1402]  INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[0:00:39.788147692] [1418]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
```

This is reproducable. That is, when the RP3 boots up from the "power cut off" state, the HQ camera is not detected. Rebooting while leaving the power connected "solves" this issue - the HQ camera is properly detected.

More info about the system: original HQ camera (actually, I tried several ones, camera standard cable, original RP power supply)

[image]

Here is the output from the raspinfo command. Note that it does claim that there are no cameras detected: 

[raspinfo.txt](https://github.com/raspberrypi/libcamera-apps/files/9900030/raspinfo.txt)

This output of raspinfo stays the same, whether the camera can be activated by libcamera-hello (after reboot)  or not (inital power on).

The camera can also be accessed after a reboot via other software like picamera2 and works fine.

I can reproduce the above described behaviour with another original HQ camera attached to the RP3 as well.

naushir, 2022-10-31:
Thank you for the report.  Could you run ``camera-bug-report`` and post the file generated by the tool here please?

cpixip, 2022-10-31:
Here we go (that has been created after a powered reboot, i.e., with the camera accessible via libcamera-apps or picamera2)

[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/9900290/bug-report.txt)

naushir, 2022-10-31:
Thank you!  Could you also possibly send me the output of ``camera-bug-report`` when you do a cold reboot and the device does not work?

cpixip, 2022-10-31:
Yes. This here is a report after the inital power on sequence:

[bug-report_Initial.txt](https://github.com/raspberrypi/libcamera-apps/files/9900342/bug-report_Initial.txt)

and this here is again the report after a powered reboot

[bug-report_reboot.txt](https://github.com/raspberrypi/libcamera-apps/files/9900348/bug-report_reboot.txt)

naushir, 2022-10-31:
@cpixip, thanks for the files!

Looking at ``[bug-report_Initial.txt](https://github.com/raspberrypi/libcamera-apps/files/9900342/bug-report_Initial.txt)``, it seems like the camera was detected correctly.  Based on your report, that boot should have not worked, is that correct?

```
--------------------------------------------------------------------------------
Cameras
--------------------------------------------------------------------------------
** libcamera-hello --list-cameras **
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]
```

Would you be able to try something to narrow down the issue please?  In ``/boot/config.txt``, can you search for ``camera_auto_detect=1`` and set that value to 0.  Then, at the end of the file, can you add ``dtoverlay=imx477``, save and exit.  Does this solve your problem after power cycle?

naushir, 2022-10-31:
> It does solve the problem. Thanks! Will you add the recipe how to solve this to the camera-forum?

Thanks for confirming.  There has been a recent change to the camera auto detection routine that has already been posted to the stable firmware - this may fix your issue. 

If you would like to try this out, run ``rpi-update`` to get the latest firmware (disclaimer: please backup anything you may need from the card first), and revert your above changes to ``/boot/config.txt``.

Let me know if that works for you.

cpixip, 2022-10-31:
Indeed `rpi-update` and reverting to the old `/boot/config.txt` did the trick as well. Thank you for this fast solution!

closed 2022-10-31 by cpixip