raspberrypi/rpicam-apps #402: Please make releases

supertobi, 2022-11-08:
Please make releases. 
Releases make it easier:
- for the user to follow development.
- for packager to create packages.
- to talk about bugs.
- to compare forks.

naushir, 2022-11-09:
Thank you for raising this issue!

All our releases happen through our public apt repos.  As such, we do not provide releases through github.  This is purely a place to provide source code and issue tracking.  However, all our apt releases are tagged with the commit sha they were build with.  You can obtain the sha by using the ``--version`` command line agrument:

```
pi@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: b89cc36c3872 19-10-2022 (13:11:40)
libcamera build: v0.0.1+21-7c855784
```

This ought to be sufficient for manual release tracking.

naushir, 2022-11-09:
Another thought, would tagging this repo when we make releases make things easer for you?

supertobi, 2022-11-09:
> Another thought, would tagging this repo when we make releases make things easer for you?

Yes, that would be wonderful. That's all I need.

naushir, 2022-11-09:
Great, I'll mark this as resolved then!

closed 2022-11-09 by naushir