raspberrypi/rpicam-apps #405: Segmenting timestamp files

ptoews, 2022-11-10:
When using `--segment` (or e.g. `--split`) it would be helpful if the timestamp file, as created by `--save-pts`, is split in the same way as the video is, so that the timestamps for each video are in an own file.

naushir, 2024-02-05:
This is a known issue.  There was some bugs in the implementation that had to be reverted (see https://github.com/raspberrypi/rpicam-apps/issues/580).  I've not had time to have a look again at this, but will do soon.

naushir, 2024-02-05:
I'm closing this as a duplicate of https://github.com/raspberrypi/rpicam-apps/issues/580.

naushir, 2025-04-01:
No plan to work on this in the near future.  For such functionality, I suggest switching to picamera2, which has support for this feature.

closed 2025-04-01 by naushir