raspberrypi/rpicam-apps #406: Slow startup of libcamera-vid with high resolutions

ptoews, 2022-11-10:
When starting a recording with a high resolution it takes a lot of time until the first frame arrives. In the following example I'm recording at 4056x3040 with 1 FPS and a timeout of 10 seconds, however I get only 3 frames. Using the verbose flag I can see that it takes a lot of time before the three frames are recorded in quick succession, implicating a startup time of ~7 seconds..

Here is the output of the bug-report tool:

[report.txt](https://github.com/raspberrypi/libcamera-apps/files/9981118/report.txt)

ptoews, 2022-11-10:
I understand that the algorithms need a few frames to converge, but does this mean that the framerate of these initial frames is coupled to the user-specified framerate? In the given example my system is able to handle e.g. 10 FPS, so if it could take the initial frames with 10 FPS it would reduce the startup time a lot. I just ran the same command with `--framerate 10` and got 91 frames, i.e. the startup time reduced to 0.9 seconds.

naushir, 2022-11-11:
> I understand that the algorithms need a few frames to converge, but does this mean that the framerate of these initial frames is coupled to the user-specified framerate?

Yes, that is correct.  

In the given example my system is able to handle e.g. 10 FPS, so if it could take the initial frames with 10 FPS it would reduce >the startup time a lot. I just ran the same command with `--framerate 10` and got 91 frames, i.e. the startup time reduced to 0.9 seconds. However, the rest of my pipeline cannot handle more than 1 FPS.

Unfortunately this is one of those things that is significantly more complicated than it seems.  Any changes to the sensor blanking time only take effect after a certain number of frames.  This number is sensor dependent.  So we would need to pipeline these changes while the algorithms converge to ensure the first output frame has the correct settings.  This in-turn limits how well and quickly the algorithms actually converge.

Secondly, if you were to, say, use > 30fps for the initial convergence, you limit the maximum shutter speed allowable during convergence, so might not get accurate results. There may be a half-way approach where we converge at 15fps (or something), but we still will get hit by the pipelining issue above.

naushir, 2022-12-01:
Unfortunately, I don't think we will be able to do anything about this, apart from disabling the drops completely (which is already under review).  I'll close this issue down as not planned.

closed 2022-12-01 by naushir

catubc, 2024-10-03:
Hi @naushir , has disabling frame drops been implemented yet?
Thanks!