raspberrypi/rpicam-apps #409: Installation

mrfoxie, 2022-11-14:
Is there anyway to install on kali linux (raspberry pi 4)?

naushir, 2022-11-15:
You might have luck compiling from the source.  Instructions can be found [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-libcamera-apps).

mrfoxie, 2022-11-15:
> You might have luck compiling from the source. Instructions can be found [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-libcamera-apps).

1libcamera-dev` package not found

naushir, 2022-11-15:
> 1libcamera-dev` package not found

Can you be more specific here please?  Did you follow the build/install instructions I linked to earlier? If yes, at what stage did it fail?

mrfoxie, 2022-11-18:
i just switched back to raspberry Pi orignal os but in future i will be working on it again till then i will close it

mrfoxie, 2022-12-10:
I tried it on kali again but this time i got error on this command
```
meson build --buildtype=release -Dpipelines=raspberrypi -Dipas=raspberrypi -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=enabled -Ddocumentation=disabled
```
Error:
```
Dependency libexif skipped: feature android disabled
Dependency libjpeg skipped: feature android disabled
Dependency libevent_pthreads skipped: feature cam disabled
Dependency libevent_pthreads skipped: feature lc-compliance disabled
Run-time dependency libtiff-4 found: YES 4.4.0
Dependency gtest skipped: feature lc-compliance disabled
Run-time dependency qt5 (modules: Core, Gui, Widgets) found: NO (tried pkgconfig)

src/apps/qcam/meson.build:4:0: ERROR: Dependency "qt5" not found, tried pkgconfig

A full log can be found at /home/kali/libcamera/build/meson-logs/meson-log.txt
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.
```

naushir, 2022-12-12:
You will need to install the qt5 libraries to get libcamera/qcam building correctly:
```
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
```

Note that this is not really a libcamera-apps specific issue.  As such, would you be able to create a new issue at https://github.com/raspberrypi/libcamera/issues if you are still having problems please?

closed 2022-12-12 by naushir