raspberrypi/rpicam-apps #414: libcamera-vid tcp stream issue

StarryDing, 2022-11-24:
When I use **libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:8088** to perform video push stream,it will be shoutdown when the client shoutdown. How to make it still work(perform video push stream) otherwise be shoutdown if the client close? 
[image]

naushir, 2022-11-25:
Can you try streaming with libav and see if that helps you?

```
libcamera-vid -t 0 --codec libav --libav-format mpegts -o "tcp://0.0.0.0:8888?listen=1"
```

naushir, 2022-12-06:
Any update on this issue?

naushir, 2022-12-07:
If there are no further updates, I will close this down shortly.

naushir, 2022-12-08:
Resolving now.  Feel free to open a new issue if you are still having problems.

closed 2022-12-08 by naushir