raspberrypi/rpicam-apps #418: [BUG] std::bad_alloc when using `--metadata` and `--codec yuv420`

ptoews, 2022-12-01:
I'm getting the following message followed by program termination when using the mentioned flags:

```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

Removing `--metadata` or choosing a different codec prevents this crash. It also seems to depend on framerate, if I set `--framerate 1` it takes longer until it happens, with default settings it happens within the first 1-2 seconds. It's also reproducible every time I try.

Here is the debugging report:
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/10132017/bug-report.txt)

naushir, 2022-12-01:
Fix is available at https://github.com/raspberrypi/libcamera-apps/pull/420

naushir, 2022-12-01:
This fix should now be merged.  We are in the process of creating a new release on apt, so this fix should find it's way there very soon!

closed 2022-12-01 by naushir

ptoews, 2022-12-01:
Awesome, you guys are quick!