raspberrypi/rpicam-apps #423: Camera is not detected

Madhuri1398, 2022-12-21:
With the Raspberry Pi 4 B model (64bit Bullseye OS) I am using OV7251 camera module of Arducam and followed the steps of building libcamera-apps [Link ](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-libcamera-apps). But I am unable to detect the camera

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/10276247/bug.txt)

6by9, 2022-12-21:
ov7251 has no official support in libcamera, so is missing a cam_helper and tuning file.
Have you added those?

Madhuri1398, 2022-12-21:
No, I haven't added them. So are you saying that cam_helper and tuning file are not available for OV7251. And because of that I am not able to detect the camera

6by9, 2022-12-21:
> So are you saying that cam_helper and tuning file are not available for OV7251. And because of that I am not able to detect the camera

Correct. Whilst the kernel drivers are present, libcamera does not have support for ov7251.

Previously covered in https://github.com/raspberrypi/libcamera-apps/issues/235. Using the cam_helper and tuning file for ov9281 seems to work, but has not been verified against datasheets or actually tuned.

Madhuri1398, 2022-12-26:
Dear Team,

Tried using OV9281 but it still not supported but now I am getting segmentation fault Can you please help me to solve this issue
[bug1.txt](https://github.com/raspberrypi/libcamera-apps/files/10301637/bug1.txt)

6by9, 2022-12-27:
> Tried using OV9281 but it still not supported but now I am getting segmentation fault

Raspberry Pi only manufacturer and officially support OV5647, IMX219, and IMX477. All other sensors are manufactured by third parties, and support should be by the manufacturer. Support from Raspberry Pi will be on an ad-hoc basis.

OV9281 does have support in libcamera.
OV7251(that this issue is about, and your log reports) does not have support, but it only requires a cam_helper and tuning file as the kernel drivers have all the required functionality for libcamera. Do note that the tuning file will now be ov7251_mono.json

libcamera is open source, so when the manufacturer doesn't bother, anyone contribute the required changes.

You've provided no detail of the seg fault. Generally that will be down to a mismatch between the libcamera library and libcamera-apps.

Madhuri1398, 2023-01-02:
Hi 
Can you please guide me to locate the path for cam_helper and tuning file for ov9281

naushir, 2023-01-03:
The tuning file for ov9281 is [here](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/raspberrypi/data/ov9281_mono.json), and the cam helper can be found [here](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/raspberrypi/cam_helper_ov9281.cpp).

I'm closing this issue down now.  The segfault you have with OV9218 is almost certainly because you are using a mix of Arducam's libcamera/ libcamera-apps binaries.  To get it working, please purge these, and install our packages from apt.  If you want support from Arducam's OV7251, please reach out to their support team.

closed 2023-01-03 by naushir

LightCannon, 2023-01-06:
I have the same issue with OV7251. I understand that the problem is that ov7251_mono.json and ov7251 helper are not found. I would like to know:
1- Where to get these 2 files
2- Where should I put them?

naushir, 2023-01-06:
You will have to speak to the vendor who supplied you with the ov7251 for this. They are welcome to provide patches the can be merged into the libcamera tree.

naushir, 2023-01-06:
There has never been native support for ov7251 in the Raspberry Pi libcamera stack or the legacy firmware stack. 

You might have been accessing the sensor directly over V4L2, which you can still do today

LightCannon, 2023-01-06:
I've been trying whole day to work over V4L2 and make it work, but no luck. My target is to use it from OpenCV and do some work after that
May you please tell me how to make ov7251  work with V4L2 and opencv can read it?
Thanks in advance.

6by9, 2023-01-06:
Use ```dtoverlay=ov7251,media-controller=0``` to disable media controller, or configure the Media Controller links using media-ctl before invoking V4L2. 
https://forums.raspberrypi.com/viewtopic.php?t=322076

LightCannon, 2023-01-06:
camera_auto_detect=0
dtoverlay=ov7251,media-controller=0

Just to confirm I understood you right, you mean this? (/boot/configs.txt) ?
```

camera_auto_detect=0
dtoverlay=ov7251,media-controller=0

```

6by9, 2023-01-06:
```camera_auto_detect=0``` will make no difference as the firmware doesn't know about ov7251
```dtoverlay=ov7251,media-controller=0``` will load the ov7251 overlay with media controller disabled (effectively reverting https://github.com/raspberrypi/linux/commit/1626186c018f3e8cb83c354a9f0b3f283855ec25)

LightCannon, 2023-01-06:
Thanks a lot. It worked well.
My issue turned out to be in "media-controller" parameter as you said.