raspberrypi/rpicam-apps #424: [Question] monochrome image sensor

fiepfiep, 2022-12-21:
Hello,

we have a monochrome image sensor for which we have made a driver.
inside the driver, we can configure multiple pixel format settings such as:
static const u32 codes[] = {
	//MEDIA_BUS_FMT_Y8_1X8,
	//MEDIA_BUS_FMT_Y10_1X10,
	//MEDIA_BUS_FMT_Y12_1X12,
	MEDIA_BUS_FMT_SGRBG8_1X8,
	MEDIA_BUS_FMT_SGRBG10_1X10,
	MEDIA_BUS_FMT_SGRBG12_1X12,

when we use the SGRBG10 variant, it works fine. (in the json camera tuning file, we have set the CCM to 0.33 for all, so it looks monochrome)
when we use the Y10 variant, we get some kind of bayer pattern in the image using libcamera-hello, although the sensor doesn't have a color filter array.
what is the correct way to do this? is there a way to completely turn off the debayering?

6by9, 2022-12-21:
After commit https://git.linuxtv.org/libcamera.git/commit/src/?id=820607b195ff942ad3533514bb2b3b2a931492d6 a mono sensor should load tuning_mono.json, whilst a colour sensor would load tuning.json.
Compare and contrast [imx296.json](https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi/data/imx296.json) and [imx296_mono.json](https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi/data/imx296_mono.json)

The ISP enumerates support for all the Bayer and greyscale formats: 
```pi@raspberrypi:~ $ v4l2-ctl --list-formats-out -d 13
ioctl: VIDIOC_ENUM_FMT
	Type: Video Output

	[0]: 'BA81' (8-bit Bayer BGBG/GRGR)
	[1]: 'RGGB' (8-bit Bayer RGRG/GBGB)
	[2]: 'GBRG' (8-bit Bayer GBGB/RGRG)
	[3]: 'GRBG' (8-bit Bayer GRGR/BGBG)
	[4]: 'pBAA' (10-bit Bayer BGBG/GRGR Packed)
	[5]: 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
	[6]: 'pGAA' (10-bit Bayer GBGB/RGRG Packed)
	[7]: 'pgAA' (10-bit Bayer GRGR/BGBG Packed)
	[8]: 'pBCC' (12-bit Bayer BGBG/GRGR Packed)
	[9]: 'pRCC' (12-bit Bayer RGRG/GBGB Packed)
	[10]: 'pGCC' (12-bit Bayer GBGB/RGRG Packed)
	[11]: 'pgCC' (12-bit Bayer GRGR/BGBG Packed)
	[12]: 'pBEE' (14-bit Bayer BGBG/GRGR Packed)
	[13]: 'pREE' (14-bit Bayer RGRG/GBGB Packed)
	[14]: 'pGEE' (14-bit Bayer GBGB/RGRG Packed)
	[15]: 'pgEE' (14-bit Bayer GRGR/BGBG Packed)
	[16]: 'BYR2' (16-bit Bayer BGBG/GRGR)
	[17]: 'RG16' (16-bit Bayer RGRG/GBGB)
	[18]: 'GB16' (16-bit Bayer GBGB/RGRG)
	[19]: 'GR16' (16-bit Bayer GRGR/BGBG)
	[20]: 'GREY' (8-bit Greyscale)
	[21]: 'Y10P' (10-bit Greyscale (MIPI Packed))
	[22]: 'Y12P' (12-bit Greyscale (MIPI Packed))
	[23]: 'Y14P' (14-bit Greyscale (MIPI Packed))
	[24]: 'Y16 ' (16-bit Greyscale)
	[25]: 'YUYV' (YUYV 4:2:2)
	[26]: 'YVYU' (YVYU 4:2:2)
	[27]: 'VYUY' (VYUY 4:2:2)
	[28]: 'UYVY' (UYVY 4:2:2)
	[29]: 'YU12' (Planar YUV 4:2:0)
	[30]: 'YV12' (Planar YVU 4:2:0)
	[31]: 'RGB3' (24-bit RGB 8-8-8)
	[32]: 'BGR3' (24-bit BGR 8-8-8)
	[33]: 'XB24' (32-bit RGBX 8-8-8-8)
	[34]: 'XR24' (32-bit BGRX 8-8-8-8)
	[35]: 'RGBP' (16-bit RGB 5-6-5)
	[36]: 'NV12' (Y/CbCr 4:2:0)
	[37]: 'NV21' (Y/CrCb 4:2:0)
	[38]: 'BG10' (10-bit Bayer BGBG/GRGR)
	[39]: 'RG10' (10-bit Bayer RGRG/GBGB)
	[40]: 'GB10' (10-bit Bayer GBGB/RGRG)
	[41]: 'BA10' (10-bit Bayer GRGR/BGBG)
	[42]: 'BG12' (12-bit Bayer BGBG/GRGR)
	[43]: 'RG12' (12-bit Bayer RGRG/GBGB)
	[44]: 'GB12' (12-bit Bayer GBGB/RGRG)
	[45]: 'BA12' (12-bit Bayer GRGR/BGBG)
	[46]: 'BG14' (14-bit Bayer BGBG/GRGR)
	[47]: 'RG14' (14-bit Bayer RGRG/GBGB)
	[48]: 'GB14' (14-bit Bayer GBGB/RGRG)
	[49]: 'GR14' (14-bit Bayer GRGR/BGBG)
	[50]: 'Y10 ' (10-bit Greyscale)
	[51]: 'Y12 ' (12-bit Greyscale)
	[52]: 'Y14 ' (14-bit Greyscale)
```
As long as your sensor advertises the correct format, libcamera should propagate that through to the ISP.
Any greyscale formats will automatically disable demosaicing, however things like lens shading still need to be programmed in all 4 channels.

fiepfiep, 2022-12-23:
Right. still, I'm not convinced that the debayering is turned off. I'd like to check the raw image, however the libcamera-apps wont let me save a DNG or PNG image when using Y10 color format. Only jpeg seems to work - but there it is hard to check the bayer pattern because of the compression. Any ideas?

pbaetens, 2023-01-02:
@davidplowman any idea to get the png/dng capture working with Y10 mono format?

davidplowman, 2023-01-16:
The first thing I would do is check that it's loading the correct mono tuning file. Please make sure you have the latest versions of everything! I normally just type some garbage in the top of the file I want it to use and check that it falls over! As was said previously, it should load a mono tuning file which should disable the de-Bayering, the AWB, the CCM etc.

As regards the DNG file, there's no case in the code for mono formats, only the Bayer ones. Someone would have to figure out what the correct values would be for a mono DNG file and then it could be added.

naushir, 2023-02-15:
As @davidplowman mentioned, until someone can list the correct values for mono formats in the DNG file, this cannot be done.
For now, I'm closing this one down due to inactivity.

closed 2023-02-15 by naushir