raspberrypi/rpicam-apps #427: [BUG] libcamera-vid does not detect all modes for ov5647

christianrauch, 2022-12-31:
**Describe the bug**
When I run `libcamera-vid --list-cameras` the camera `ov5647` is detected with the 4 modes:
```
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
```

The [OV5647 datasheet](https://docs.rs-online.com/6580/0900766b811d1155.pdf) lists more modes under "Product Specifications":
> maximum image transfer rate:
> - QSXGA (2592x1944): 15 fps
> - 1080p: 30 fps
> - 960p: 45 fps
> - 720p: 60 fps
> - VGA (640x480): 90 fps
> - QVGA (320x240): 120 fps

The modes with VGA @ 90 fps and QVGA @ 120 fps are not detected by `libcamera-vid`.

**Bug report**
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/10327323/bug-report.txt)

6by9, 2022-12-31:
The product brief from Omnivision documents potential capabilities of the sensor, not what is implemented in any particular driver.

The QVGA modes have never been implemented in the firmware drivers or kernel.

I thought that the VGA mode went up to 90fps happily, not the 58.92fps reported by libcamera-apps. That likely implies there is something incorrectly reported by the kernel driver in that mode, but it won't be investigated until after the holiday break.

christianrauch, 2023-01-01:
Can it be that this is because there are simply only 4 modes in the [OV5647 Linux driver](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ov5647.c?h=v6.2-rc1#n492)?

naushir, 2023-01-03:
> Can it be that this is because there are simply only 4 modes in the [OV5647 Linux driver](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/media/i2c/ov5647.c?h=v6.2-rc1#n492)?

That's correct, the modes listed through ``libcamera-vid --list-cameras`` simply report the modes advertised by the kernel driver.

christianrauch, 2023-01-03:
I experimented with the ov5647 kernel module and created an out-of-tree fork at https://github.com/christianrauch/ov5647. This fork contains the Raspberry Pi specific commits from https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/media/i2c/ov5647.c and reverts the commit that removed the SBGGR8 mode (["media: ov5647: Remove 640x480 SBGGR8 mode"](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=38c22308181562d52926b6701327c3fe209f6d84)). With this out-of-tree module, I get a new mode that enables framerates up to 80Hz.

The upstream ov5647 module without the Raspberry Pi specific commits won't work on the Raspberry Pi. Specifically, I needed ["media: i2c: ov5647: Add support for regulator control."](https://github.com/raspberrypi/linux/commit/0a1445dde5870797cdd8c19471ad3171ae0dcb41) in order to connect to the camera. Is there a reason why those Raspberry Pi commits are not upstreamed?

naushir, 2023-01-10:
Given this is a kernel driver change and not libcamera-apps, I'll close this down over here.

closed 2023-01-10 by naushir