raspberrypi/rpicam-apps #436: Raspberry Pi 3 B+ & IM708

ApolloRR, 2023-01-12:
The New Raspberry Pi Camera Module 3 does not work with the Raspberry Pi 3 B+.

`Preview window unavailable
[0:01:02.657634468] [1181]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
ERROR: *** no cameras available ***`

I've so far attempted to modify the `\boot\config.txt` by setting `dtoverlay=imx708` and `camera_auto_detect=0`.

`dtoverlay=imx708` was added once in the `# Additional overlays and parameters are documented /boot/overlays/README` section, and also at the very botton of the `\boot\config.txt` file.

`pi@lidar:~ $ vcgencmd version
Aug 26 2022 14:04:36
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)`

Bug-report.txt attached:
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/10403483/bug-report.txt)

davidplowman, 2023-01-12:
Hi, thanks for attaching the bug report. That's showing that `uname -a`, `vcgencmd version` and `libcamera-hello --version` are all reporting dates from last August, which is much too old for the IMX708 to work. Can you check if
```
sudo apt update
sudo apt full-upgrade
```
wants to update anything? After doing that myself, all three commands are reporting dates from earlier this January. Thanks!

ApolloRR, 2023-01-12:
Whoops! I had mistakenly thought the latest image already contained the update.

After the upgrade and restart the camera works.

closed 2023-01-12 by ApolloRR