raspberrypi/rpicam-apps #439: Preview window error even if choosing not to have preview window

rafael2k, 2023-01-15:
I build libcamera from today's git with QT and X11 support on Debian stable, arm64.

Running (or any other app with --nopreview or -n):
libcamera-jpeg  --width 1280 --height 720 --nopreview  --camera 1 -o out.jpg

I get:
```
(...)
[2:53:39.495994549] [8866]  INFO Camera camera.cpp:1026 configuring streams: (0) 1920x1080-SBGGR8
terminate called after throwing an instance of 'std::runtime_error'
  what():  Preview windows only support YUV420
Aborted
```
Hardware is a pinephone with ov5640 camera (and a galaxy core too, but the error happens with both).

Thanks,
Rafael Diniz

naushir, 2023-01-16:
@rafael2k thank you for the feedback.

The Raspberry Pi libcamera-apps are only really supported on the Raspberry Pi platform.  Specifically, we only ever use YUV420 image formats for still image and video encoding.  You can get around this particular exception with the following patch:

```diff
diff --git a/core/libcamera_app.cpp b/core/libcamera_app.cpp
index 04696fb6f4ac..dbb37c976abf 100644
--- a/core/libcamera_app.cpp
+++ b/core/libcamera_app.cpp
@@ -951,7 +951,7 @@ void LibcameraApp::previewThread()
                                preview_cond_var_.wait(lock);
                }

-               if (item.stream->configuration().pixelFormat != libcamera::formats::YUV420)
+               if (!options_->nopreview && item.stream->configuration().pixelFormat != libcamera::formats::YUV420)
                        throw std::runtime_error("Preview windows only support YUV420");

                StreamInfo info = GetStreamInfo(item.stream);
```

but you are going to hit other points in the code that will expect the image buffer to be in YUV420 format.

One possible approach is to get the pipeline handler for the SoC on the pinephone to explicitly output YUV420.  I am not sure if the ISP hardware can do this or not.

rafael2k, 2023-01-17:
Thanks, will push it to my fork. Could this be PR to upstream?

naushir, 2023-01-17:
If you find this works for you, feel free to create a PR and we can merge it.  However, as mentioned earlier, I think there will be loads of other places in libcamera-apps that expects YUV420 format, and most things still won't function correctly.

naushir, 2023-01-18:
If there is nothing further required for this issue, feel free to close this down.

naushir, 2023-01-18:
> @naushir is the goal of this project to only support the Raspberry Pi or is the plan to support other platforms / pipeline handlers in the future?

The goal of ``libcamera-apps`` is to provide a (mostly) like-for-like replacement for our legacy firmware based camera applications.  As such, we have written the applications with only the Raspberry Pi platform in mind.

Regarding the YUY420 format restriction - if the community find benefit to support more formats than YUV420, we will be happy to review PRs for this work.  However, it is not something our very small team will be able to work on directly.   This is nothing specifically other than this that blocks the use of other pipeline handlers running under libcamera-apps.  But note, that we will not actively test other platforms like we do on Raspberry Pi.

> If the former, maybe it should be renamed to `rpi-libcamera-apps` or something along those lines to avoid confusion and don't give the wrong impression that is a generic libcamera application?

We have written vast streams of documentation/guides/tutorials for our userbase were this suite is called ``libcamera-apps``.  Unfortunately, changing this now would be unfeasible.  Given that this project is in the Raspberry Pi GitHub repo, and the only way to get the deb packages would be through the Raspberry Pi apt repo, I would hope this does not cause too much confusion.  Perhaps a more explicit message in the documentation may be warranted?

naushir, 2023-01-18:
I just want to add something to further clarify the above point.

We use standard Linux APIs for pretty much everything in libcamera-apps.  The libcamera interface is standard, the V4L2 M2M codec interface is standard (although the device node may be hardcoded for the pi kernel, not sure, the DRM/QT/EGL display framework interfaces are standard, libjpeg encoder is standard.

The only restriction is that we currently expect YUV420 format frames because that is standard format for still image and video codec operations.  There may be one or two minor things that may be Raspberry Pi specific (nothing comes to mind), but they will be very minor.

If folks would like to get these apps to run on other platforms, it ought to be possible provided the platform implements these libraries according to the published standard APIs.

naushir, 2023-01-23:
I'll be closing this down now as there is no active work being done on this.  Feel free to re-open this or new issues if you have any other questions.

closed 2023-01-23 by naushir