raspberrypi/rpicam-apps #440: libcamera-apps crash on Pi zero/bullseye with the new pi camera v3

duhanebel, 2023-01-16:
**Describe the bug**
After compiling from scratch both libcamera (https://github.com/raspberrypi/libcamera.git@5df5b7) and libcamera-apps ( https://github.com/raspberrypi/libcamera-apps.git@29f394) on the PiZero, I'm unable to use any of the apps to access the camera. Every app crashes with the message: `FATAL Serializer control_serializer.cpp:598 A list of V4L2 controls requires an ControlInfoMap`

**Bug report**
The bug report tool uses the following syntax:

```
camera-bug-report -t 0 -o bug.txt -c "lib camera-still -o test.jpg"

--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 16-01-2023 (16:33:30)
Command: /usr/local/bin/camera-bug-report -t 0 -o bug.txt -c libcamera-still -o test.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 0000000054148f67
Model           : Raspberry Pi Zero W Rev 1.1

--------------------------------------------------------------------------------
Configuration
Logs
--------------------------------------------------------------------------------
** dmesg **
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.87+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1616 Fri Jan 13 16:55:19 GMT 2023
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x07c00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000017ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000017ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000017ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 97440
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i,margin_left=32,margin_right=32,margin_top=32,margin_bottom=32 smsc95xx.macaddr=B8:27:EB:14:8F:67 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=1ef17acc-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order:
[    0.142665] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.142998] Serial: AMBA PL011 UART driver
[    0.157569] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.191092] raspberrypi-firmware soc:firmware: Attached to firmware from 2023-01-13T16:01:22, variant start
[    0.201131] raspberrypi-firmware soc:firmware: Firmware hash is 8e24320e396bce8f9177405e514770bb8b47473f
[    0.261003] Kprobes globally optimized
[    0.272991] bcm2835-dma 20007000.dma: DMA legacy API manager, dmachans=0x1
[... 289 lines cut ...]
??? [0xffffffffb6eeb300] (/usr/lib/arm-linux-gnueabihf/ld-2.31.so [0xffffffffb6eeb300])

```

**Additional context**
The camera is recognised properly:

```
$ cam -l
[5:52:07.644457846] [6962]  WARN IPAManager ipa_manager.cpp:114 Public key not valid
[5:52:07.651198746] [6962]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+55-5df5b72c
[5:52:07.960697157] [6963]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media0
Available cameras:
1: 'imx708' (/base/soc/i2c0mux/i2c@1/imx708@1a)
```
and
```
$ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx708 [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```

naushir, 2023-01-17:
The issue you see is likely related to this log message:

>```
> $ cam -l
> [5:52:07.644457846] [6962]  WARN IPAManager ipa_manager.cpp:114 Public key not valid
>```

The IPA is running in an isolated process because of a public/private key verification failure.  I know this mode of operations had a few issues working correctly recently, it's likely not fixed entirely.

To stop using this isolated mode, make sure you have the tls/ssl libraries installed
```
sudo apt install -y libgnutls28-dev openssl
```
and do a fresh meson configure/build step.  That ought to fix your issue.

naushir, 2023-01-17:
@kbingham - I've not tried IPA isolation mode recently, do you know of any outstanding issues with it?

duhanebel, 2023-01-17:
Ah! I didn't realise I've built it with IPA isolation. Let me try without and see if the issue improves!

naushir, 2023-01-17:
Just tried a test with forcing IPA isolation:
```
LIBCAMERA_IPA_FORCE_ISOLATION=1 libcamera-hello -t 0 --qt-preview
```
and things did seem to work correctly for me.  Not sure if your issue may be related to some build artefacts or not now...

kbingham, 2023-01-17:
Yes, as far as I'm aware isolation should still work. I'll make sure it gets on the automated tests here on the RPi device.

duhanebel, 2023-01-17:
I confirm it works without isolation 👍