raspberrypi/rpicam-apps #442: [BUG] Failed to load suitable IPA Library for IMX708

brianabouchard, 2023-01-23:
**Describe the bug**
I've been able to successfully get my pi camera 3 (IMX708) to load on Ubuntu 22.04 with the linux-raspi kernel 6.2. The cam debug tool in libcamera is able to display and save images, but the libcamera-apps are not. Apps all fail with "Failed to load suitable IPA Library for IMX708.

**Bug report**
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/10476909/bug-report.txt)

**Additional context**
I cloned and built the "next" branch noting that several IMX708 additions had been made there but that did not resolve the issue as I had hoped.

naushir, 2023-01-23:
Thank you for reporting the issue!

It looks like this may be an out-of-date libcamera library install that is causing your issue. Support for imx708 is not yet available in the upstream libcamera tree.  We have sent the patches to be reviewed, so hopefully this should not be too long now.

In the meantime, you must build the libcamera libraries from our downstream libcamera tree at https://github.com/raspberrypi/libcamera/tree/main.  This has all the support needed for the imx708 sensor.

williamhobbs, 2023-01-24:
I'm not sure if this is related, but I've been able to use the IMX708 (Camera Module 3, standard lens, NoIR) with "stock" libcamera from a fresh Raspberry Pi OS install (I think I ran a full update). 

I did notice that I can't get an exposure longer than 1.72 seconds, though, which seemed odd.

brianabouchard, 2023-01-24:
> EDIT: I missed that you seem to have `cam` working correctly, so that is a bit confusing. `cam` and `libcamera-apps` use exactly the same IPA, so if one works the other really ought to. Can you clarify how/where you built libcamera from?

Yes, this struck me as odd as well! I did first build libcamera from source with the downstream tree and that wasn't working right, which is why I tried the upstream tree. Perhaps when I wiped that initial install the IPA stuck around? Not exactly sure, but I am going to try a clean ubuntu install with a clean libcamera and libcamera-apps install to see if that works and will update you with what I find.

naushir, 2023-01-25:
> I did notice that I can't get an exposure longer than 1.72 seconds, though, which seemed odd.

This is because the long exposure feature did not get integrated into the kernel driver in time for the product launch.  This has since been merged https://github.com/raspberrypi/linux/pull/5306, so newer kernel builds will show a longer maximum exposure time.

naushir, 2023-01-25:
> Yes, this struck me as odd as well! I did first build libcamera from source with the downstream tree and that wasn't working right, which is why I tried the upstream tree. Perhaps when I wiped that [...]

That would be a useful thing to try.  The upstream tree will definitely not work with the imx708 until the patches have been merged.

naushir, 2023-02-03:
Is there anything more to look at here?  If not, I'll close this issue down.

naushir, 2023-02-06:
I'll close this one down for now.  Please do re-open if you are still seeing issues.

closed 2023-02-06 by naushir