raspberrypi/rpicam-apps #445: Feature request: --info-text command line output

williamhobbs, 2023-01-24:
I'm not sure if feature requests are allowed, or if this is the correct place, so please let me know if this should be redirected somewhere else.

It seems like there would be value in having an option to output --info-text data to the command line for configurations where there is no desktop/windowed environment available. 

In particular, being able to focus a lens with something like:

`libcamera-hello --info-test "Focus measure: %focus`

that outputs to the terminal would be helpful.

davidplowman, 2023-01-25:
As a workaround you might try using _Picamera2_, which is often a good solution where people have specific requirements that would otherwise add yet more options to these apps. For example:
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start(show_preview=True)
while True:
    print(picam2.capture_metadata()['FocusFoM'])
```
Does that help at all?

naushir, 2023-01-25:
As @davidplowman said, python is probably the easiest place to implement such custom application behaviour.

However, it turns out that this is actually quite easy to plumb in to libcamera-apps.  I've raised a PR at https://github.com/raspberrypi/libcamera-apps/pull/446.

naushir, 2023-01-25:
https://github.com/raspberrypi/libcamera-apps/pull/446 has now been merged.

closed 2023-01-25 by naushir