raspberrypi/rpicam-apps #448: compilation error

Abu-Abdullah, 2023-01-28:
Hi,

i'm getting this error while compiling from src:
```
[ 55%] Built target libcamera_app
[ 57%] Building CXX object encoder/CMakeFiles/encoders.dir/libav_encoder.cpp.o
/home/pi/libcamera-apps/encoder/libav_encoder.cpp: In member function 'void LibAvEncoder::initAudioOutCodec(const VideoOptions*, const StreamInfo&)':
/home/pi/libcamera-apps/encoder/libav_encoder.cpp:189:24: error: 'AVCodecContext::channels' is deprecated [-Werror=deprecated-declarations]
  189 |  codec_ctx_[AudioOut]->channels = stream_[AudioIn]->codecpar->channels;
      |                        ^~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
make[2]: *** [encoder/CMakeFiles/encoders.dir/build.make:134: encoder/CMakeFiles/encoders.dir/libav_encoder.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:326: encoder/CMakeFiles/encoders.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

naushir, 2023-01-31:
Thank you for the report!  Can you confirm where you installed the ffmpeg dev libraries from?

marcusfolkesson, 2023-01-31:
I think AVCodecContext::channels is listed as deprecated from ffmpeg 5.1.
I have this error as well.

[12:20:00]marcus@goliat:~$ ffmpeg -version
ffmpeg version n5.1.2 Copyright (c) 2000-2022 the FFmpeg developers
built with gcc 12.2.0 (GCC)
configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
libavutil      57. 28.100 / 57. 28.100
libavcodec     59. 37.100 / 59. 37.100
libavformat    59. 27.100 / 59. 27.100
libavdevice    59.  7.100 / 59.  7.100
libavfilter     8. 44.100 /  8. 44.100
libswscale      6.  7.100 /  6.  7.100
libswresample   4.  7.100 /  4.  7.100
libpostproc    56.  6.100 / 56.  6.100

naushir, 2023-01-31:
Ah, that explains things.  libcamera-apps uses the Raspberry Pi builds of ffmpeg that are pre v5.1.  It's worth noting that the Raspberry Pi ffmpeg builds have extensions for zero copy hardware encode operations.  So the libav codec in libcamera-apps will not work with native upstream ffmpeg.

maknon, 2023-01-31:
> Thank you for the report! Can you confirm where you installed the ffmpeg dev libraries from?

sorry, i forgot to mentioned this. yes i compiled everything manually with latest releases including opencv, ffmpeg, gst and the rest as i need some time to test new features of gst. ffmpeg is compiled with 5.12

now im not sure how to compile it manually and at the same time preserve the "zero copy hw encode"

im compiling ffmpeg with this (for both old/new stack)
```
sudo -E ./configure \
 --extra-ldflags='-L/opt/vc/lib' \
 --extra_cflags='-Ofast -ffast-math -funsafe-math-optimizations -lm -Wno-error -Wno-redundant-decls -I/opt/vc/include/IL -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux' \
 --enable-cross-compile \
 --arch=armel \
 --target-os=linux \
 --enable-shared \
 --disable-static \
 --enable-gpl \
 --enable-libx264 \
 --disable-libopenmpt \
 --disable-neon \
 --enable-omx \
 --enable-nonfree \
 --enable-omx-rpi \
 --enable-mmal \
 --disable-doc \
 --disable-postproc
```

is there away ?

Abu-Abdullah, 2023-01-31:
sorry, i forgot to mentioned this. yes i compiled everything manually with latest releases including opencv, ffmpeg, gst and the rest as i need some time to test new features of gst. ffmpeg is compiled with 5.12

now im not sure how to compile it manually and at the same time preserve the "zero copy hw encode"

im compiling ffmpeg with this (for both old/new stack)
```
sudo -E ./configure \
 --extra-ldflags='-L/opt/vc/lib' \
 --extra_cflags='-Ofast -ffast-math -funsafe-math-optimizations -lm -Wno-error -Wno-redundant-decls -I/opt/vc/include/IL -I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux' \
 --enable-cross-compile \
 --arch=armel \
 --target-os=linux \
 --enable-shared \
 --disable-static \
 --enable-gpl \
 --enable-libx264 \
 --disable-libopenmpt \
 --disable-neon \
 --enable-omx \
 --enable-nonfree \
 --enable-omx-rpi \
 --enable-mmal \
 --disable-doc \
 --disable-postproc
```
is there away ?

naushir, 2023-01-31:
Is there any specific reason that you are compiling ffmpeg yourself?  Instead, would you be able to use our pre-compiled libraries available through ``apt``?  The latter would have all our custom downstream changes needed for the encode path.

Abu-Abdullah, 2023-01-31:
ok i can go with this as well. gst is my target not the ffmpeg. thanks for the support.

closed 2023-01-31 by naushir