raspberrypi/rpicam-apps #449: Feature Request: Multiple displays drm_preview

schoolpost, 2023-01-28:
ShowPreview on multiple displays using the drm_preview. 

currently only supports 1, would be nice to have at least have 2. 1 for an HDMI monitor and another over a DSI / DPI attached LCD for example.

naushir, 2023-02-10:
I'm afraid I know very very little about DRM. @6by9, do you think this is feasable?

6by9, 2023-02-10:
You can assign the same buffer to multiple planes, with each plane associated with a different display (crtc & connector) - this is actually what X does as it renders to one "extended desktop" buffer, and then gives each display a source rectangle to control which bits they show.

It does make life a little more interesting when it comes to returning buffers if the displays are running at different refresh rates - I think refresh will effectively be limited to the lowest refresh rate. (I've just tried ```kmstest --flip``` with one 1280x800@60 and one 4k30 screen, and I get 30fps rendering).

My understanding is that you want to do an atomic commit of multiple planes at the same time. That will end up synchronised to the lower refresh rate unless you also use ```DRM_MODE_ATOMIC_NONBLOCK``` and/or ```DRM_MODE_PAGE_FLIP_ASYNC```
It looks like kmsxx is supporting atomic commits and probably makes a reasonable example - https://github.com/tomba/kmsxx/blob/master/kms%2B%2B/src/atomicreq.cpp

naushir, 2024-09-10:
Closing due to inactivity.

closed 2024-09-10 by naushir