raspberrypi/rpicam-apps #463: "No cameras detected" with multiple hardware combos [BUG]

SlickStretch, 2023-02-10:
Hello everybody. I've been struggling with this for a long time, I hope you can help me.

I can't get my cameras working.

Here's some background information:

-I have 2 Raspberry Pi's: A Pi Zero W, and a Pi 3b.
-I have 2 ribbon cables for the Pi 3b, and 1 ribbon cable for the Pi Zero W, all of which I have tested with my multimeter.
-I have 2 Pi Camera modules ver. 1.3. One is brand new, and one worked fine when I put it away a couple of years ago.

I have tried every combination of the above hardware, and all give me the same following result.

-I have tried the following operating systems on the Pi 3: Raspberry Pi OS Lite, Raspberry Pi OS, and OctoPi 1.0.0rc (End Goal)
-I have tried the following operating systems on the Pi Zero: Raspberry Pi OS Lite, Raspberry Pi OS

Here's the current OS version:

```
pi@octopi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

Here's what happens when I try to take a picture:
```
pi@octopi:~ $ libcamera-hello
Preview window unavailable
[0:27:15.551839109] [7160]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
ERROR: *** no cameras available ***
```

Here's the libcamera bug report:
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 08-02-2023 (01:34:58)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
octopi

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
[... 838 lines cut ...]
Linux octopi 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

** vcgencmd version **
Jan  5 2023 10:49:28 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start_x)

** libcamera-hello --version **
libcamera-apps build: 9f08463997b8 26-01-2023 (16:29:53)
libcamera build: v0.0.3+40-9b860a66
```

Any help is appreciated, thank you!

6by9, 2023-02-10:
> I have 2 Pi Camera modules ver. 1.3. One is brand new, 

Raspberry Pi stopped selling the V1.3 camera in about 2016 when the v2 camera out. OV5647 was meant to have been end of life at that point. If you've just bought an OV5647 sensor, it wasn't from Raspberry Pi.

In your config.txt
```
# Automatically load overlays for detected cameras
start_x=1
...
# enable raspicam
start_x=1
gpu_mem=128
```
These are not the default settings in Bullseye and imply you have enabled the legacy camera stack. Don't do that if you want to use libcamera.

In vcdbg log msg
```
012461.630: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
012757.271: camsubs: Camera not found
012757.431: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
012758.146: camsubs: Camera not found
012758.289: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
012759.002: camsubs: Camera not found
012759.142: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
012759.346: camsubs: Camera not found
012759.489: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 134, power enable gpio = 133
012760.203: camsubs: Camera not found
```
The firmware has done its best to find the camera, but has found nothing. Generally that would point towards a hardware issue. Are you sure that you have the camera connections the right way up, and on the 3B you've used the camera connector and not the display one?
Most V1.3 camera modules had an LED connected. Does that blink during boot as the firmware tries probing? (On your current setup that would appear to be about 12 seconds in)

SlickStretch, 2023-02-11:
Sorry for the late response, I've been very busy lately.

> These are not the default settings in Bullseye and imply you have enabled the legacy camera stack. Don't do that if you want to use libcamera.

I enabled the legacy stack while trying to get it working. It wasn't enabled at first, and enabling it didn't seem to make any difference.

> The firmware has done its best to find the camera, but has found nothing. Generally that would point towards a hardware issue.

Yes, but I have a hard time believing that a hardware issue will act the exact same accross 2 pi's, 3 cables, and 2 cameras. Is that likely? If this is the problem, any suggestions as to how I can figure out which piece of hardware is broken?

> Are you sure that you have the camera connections the right way up, and on the 3B you've used the camera connector and not the display one?

Yes. I double-checked the connections against this page: https://projects.raspberrypi.org/en/projects/getting-started-with-picamera/2

> Most V1.3 camera modules had an LED connected. Does that blink during boot as the firmware tries probing?

There is an LED. It does not blink.

Thank you for taking the time to help me.

SlickStretch, 2023-02-11:
Sorry for the late reply. I've been very busy lately.
 
 > These are not the default settings in Bullseye and imply you have enabled the legacy camera stack. Don't do that if you want to use libcamera.
 
 I was already having this problem before I enabled it when trying to get it working. Enabling it didn't seem to make any difference.

> The firmware has done its best to find the camera, but has found nothing. Generally that would point towards a hardware issue.

Agreed. But, I have a hard time believing that a hardware issue would persist accross 2 pi's, 3 ribbon cables, and 2 cameras. I can switch out the pi, cable, and camera, leaving **none** of the same hardware, and I still have the same problem.

> Are you sure that you have the camera connections the right way up, and on the 3B you've used the camera connector and not the display one?

Yeah. I double-checked using [this web page.](https://projects.raspberrypi.org/en/projects/getting-started-with-picamera/2)

> Most V1.3 camera modules had an LED connected. Does that blink during boot as the firmware tries probing?

Yeah, both cameras have an LED. They do nothing during boot.

Thanks for taking the time to help me.

SlickStretch, 2023-02-14:
Unfortunately I don't, and I don't want to buy another one right now. I'm just going to use a USB webcam for now. Thanks for your time.

closed 2023-02-14 by SlickStretch