raspberrypi/rpicam-apps #466: DRM preview unavailable - not master

RagdollCar, 2023-02-11:
I have attached a new LCD screen to my RPI4:
https://botland.com.pl/wyswietlacze-raspberry-pi/5337-ekran-dotykowy-pojemnosciowy-lcd-ips-7-c-1024x600px-hdmi-usb-rev-41-do-raspberry-pi-obudowa-czarno-biala-waveshare-11303-5904422371340.html

Connecting this screen required adding to `/boot/config.txt`:

```
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

Then I tried to show frames from my IMX477 on such LCD screen but I ended up with an exception thrown from drm_preview.cpp: `DRM preview unavailable - not master`.

I disconnected this LCD screen and switched to my monitor but I also ended up with `DRM preview unavailable - not master`. This exception was not thrown until I tried to use another LCD screen.

Things I did e.g.:
- reverted modifying `/boot/config.txt`
- rebuilding libcamera and libcamera-apps
- rebooting of course

It seems like another device/process became DRM master and I cannot preview frames from the camera?

6by9, 2023-02-12:
```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```
Are all largely irrelevant if using vc4-kms-v3d (the default on Bullseye).
Your screen is HDMI, so as long as it has a valid EDID, vc4-kms-v3d should adopt that resolution. You may need to add ```disable_fw_kms_setup=1``` to stop the firmware trying to persuade KMS what resolution to use.
Waveshare's instructions will often tell you to remove ```dtoverlay=vc4-kms-v3d``` from config.txt. This is bad advice and would likely cause issues to those you are seeing.

What environment are you using? Raspberry Pi OS with desktop, or Lite / booting to CLI?

DRM only supports a single master at any one time, therefore if you have Xserver or other window manager running then it will claim that role and stop libcamera-apps from requesting it.
Trying to run two instances of libcamera-[still|vid|hello] on a CM will also cause that error, as the first one will claim the master role and hence block the second.

RagdollCar, 2023-02-12:
Adding `disable_fw_kms_setup=1` instead of

```
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0
hdmi_drive=1
```

also allows the LCD screen to work, so thanks for this tip.

But the initial problem with DRM remains. How can I check which process is current DRM master so maybe I can try to kill it?

I am using RPI OS with desktop:
```
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 11 (bullseye
Release: 11
Codename: bullseye
```

RagdollCar, 2023-02-12:
Okay, I fixed it when reading once again that Window Manager might be the master of DRM. I switched to terminal mode and tried to run libcamera-hello and it started previewing frames.

But the preview was working even when I was using desktop mode, so it's strange.

6by9, 2023-02-12:
> But the preview was working even when I was using desktop mode, so it's strange.

You hadn't said that the preview worked before.

libcamera-apps have 3 different routes for rendering - EGL, DRM, and QT.
- EGL requires Glamour and an X environment
- DRM needs to be able to acquire the master role, so no X.
- QT is the worst performing, but requires the least from the installed environment.

At a guess you've managed to disable EGL rendering and it's falling back to QT, but I could be totally wrong there.

RagdollCar, 2023-02-12:
I am using only DRM. Command `cmake` prints that EGL and QT is disabled.

RagdollCar, 2023-02-12:
I am closing becuase the solution was to switch to terminal mode (Ctrl + Alt + F1) and launch previewing without Window Manager.

closed 2023-02-12 by RagdollCar