raspberrypi/rpicam-apps #470: Frame drops while recording long videos at high frame rates

GuyPerets106, 2023-02-14:
Hi, I'm working on a project which requires high frame rate video (120fps or higher). 
My specs vary from a few RPi 4B's 8GB Ram, With a fast SD card to Waveshare Mini IO boards + CM4's 8GB Ram 16GB eMMC. Both run the latest Bullseye 64bit.  
My cameras are Waveshare OV9281-110 & 120 which are capable shooting high frame rates.

I noticed that above 20 seconds of recording I get frame skips with libcamera-raw and libcamera-vid can hold up to 40 seconds without dropping frames. There's a command which can help me never skip any frame? It's really important.

Thank you!

naushir, 2023-02-14:
Can you please provide the exact command lines you are using for the encode operation please?

We have some hints for fast framerate captures in our documentation [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#high-framerate-capture)

GuyPerets106, 2023-02-14:
> 

I am using:
_libcamera-raw --mode 640:400:8:U --framerate 120 --shutter 500 --gain 5 --denoise cdn_off -o test.raw_ 
with _-t 45000_ for example (which resulting a 29-30 seconds video for some reason) , and frame drops.
I also tried _--frames (120*seconds)_ but I found out it worked only on libcamera-vid and not on raw sadly.

I will try to follow the instructions you shared , and see if this helps. Thank you

naushir, 2023-02-14:
With libcamera-raw, the bottleneck is likely going to be the write speed on the sdcard.  The instructions provided in my last comment are more targeted to libcamera-vid.  Some further questions:

- Do you definitely need RAW output?
- What sensor mode/resolution is being used?
- Would it be possible to capture the output to a RAM disk to see if that helps?

You could possibly try setting ``--buffer-count 12`` to see if that helps with the frame drops in libcamera-raw.

6by9, 2023-02-14:
640x480 as YUV420 is 450kB. At 120fps that is then just under 53Mbytes/sec (420Mbit/s).
Storage will be the bottle-neck. You mention both SD card and CM4 eMMC - which are you actually using?

(Just beaten to it by naushir!)

GuyPerets106, 2023-02-14:
> With libcamera-raw, the bottleneck is likely going to be the write speed on the sdcard. The instructions provided in my last comment are more targeted to libcamera-vid. Some further questions:
> 
> * Do you definitely need RAW output?
> * What sensor mode/resolution is being used? You can tell by searching the libcamera-raw output for the string `Selected sensor format:`.
> * Would it be possible to capture the output to a RAM disk to see if that helps?
> 
> You could possibly try setting `--buffer-count 12` to see if that helps with the frame drops in libcamera-raw.

I need RAW output because I compress with ffmpeg and need to control the bitrate and filter out noise myself. If you know libcamera-vid could do this and in a better way it'll be great. 
`Selected sensor format: 640x400-Y8_1X8`
It would be possible to write to the RAM , but I need long videos and how can I make the data continously write itself from the RAM to the SD card without the memory get full? 
I also ran a `dd` test to see the speeds of my SD card (RPi 4B) and got around 35MB/s Write | 45MB/s Read. 
The eMMC CM4 got 68MB/s Write | 85MB/s Read.

> 640x480 as YUV420 is 450kB. At 120fps that is then just under 53Mbytes/sec (420Mbit/s). Storage will be the bottle-neck. You mention both SD card and CM4 eMMC - which are you actually using?
> 
> (Just beaten to it by naushir!)

I will eventually be using the CM4 with eMMC , but for now I'm running tests with 4B's.
Notice I currently doing 640x400 and not 480. You're basically saying that if I will ever need to record in higher resolution (1280x720/800) it won't even be possible for long videos?

naushir, 2023-02-14:
> I need RAW output because I compress with ffmpeg and need to control the bitrate and filter out noise myself. If you know libcamera-vid could do this and in a better way it'll be great.

It might be the case that libcamera-vid does what you want and much more efficiently because it does not store RAW frames.

Can you elaborate on your application a bit more? How are you using ffmpeg to compress the stream, and with what codec?  How is this denoise operation performed?  Are you doing any other interim image processing before compression?

6by9, 2023-02-14:
My numbers are off as saving the Y8 raw from the sensor will be a 33% saving over YUV420 as it has no chroma planes. It's still a lot of data.

SDcard or EMMC storage is almost always slow in comparison to raw video streams. They also need to pause if they ever need to erase cells, so you will get random delays particularly as they age.
Seeing as you're looking at Compute Modules, you could add an NVMe module to the PCIe slot for faster storage.

1280x800 @ 120fps Y8 is going to be 120MBytes/s, or nearly 1Gbit/s. That's certainly too much for SD cards or EMMC over the SDIO interface. NVMe or USB3 flash drive may have a chance.

GuyPerets106, 2023-02-14:
> It might be the case that libcamera-vid does what you want and much more efficiently because it does not store RAW frames.
>
> Can you elaborate on your application a bit more? How are you using ffmpeg [...]

The videos I eventually use are regular '.mp4' files. I can't really elaborate about the application itself , but I'll say I tried libcamera-raw + ffmpeg for compression AND libcamera-vid and got two different videos brightness-wise and quality-wise. I still need to figure out if the libcamera-vid will be good to my needs , but it'll take time so now I just test with both raw and vid. **My main concern is stable framerate** , and I don't know how to get it using libcamera-raw/vid - both are around 100-110fps when I specifically use `--framerate 120` and `--frames (120*seconds)` with vid or `-t 1000*seconds` with raw.

> My numbers are off as saving the Y8 raw from the sensor will be a 33% saving over YUV420 as it has no chroma planes. It's still a lot of data.
>
> SDcard or EMMC storage is almost always slow in [...]

NVMe is something I tried before and helped me. But something changed when I started to use libcamera - I couldn't get stable framerate , in any mode and any length. Like the prompt about framerate I write in the command line are more like suggestion , even when I take 1/2/5sec videos.

6by9, 2023-02-14:
You still haven't given any detail as to what you are trying to achieve.

Do you need AE/AGC, or is it controlled lighting conditions? You appear to have fixed gain and shutter, so presumably controlled. In which case you don't need libcamera, so drop back to raw V4L2 and capturing frames that way.

You either need to use ```dtoverlay=ov9281,media-controller=0``` to allow resolution selection via /dev/video0 (eg ```v4l2-ctl -v width=640,height=400,pixelformat=GREY```), or issue the relevant media-ctl runes to set up the pipeline.
```v4l2-ctl --stream-mmap=3 --stream-to=file.y8 --stream-count=12000``` should then save 100seconds of raw stream to file.y8, and it should print out a frame rate once a second. If that can't keep up, then you've a lost cause. (Writing to /dev/null will work at full speed).

You'll need to compute your own values for V4L2_CID_EXPOSURE, V4L2_CID_ANALOGUE_GAIN, and V4L2_CID_VBLANK, or run libcamera with the relevant parameters and then use ```v4l2-ctl -d /dev/v4l-subdev0 --list-ctrls``` to print out the values that libcamera computes for you. (Note that libcamera requires media controller to be enabled).

GuyPerets106, 2023-02-15:
> You still haven't given any detail as to what you are trying to achieve.
>
> Do you need AE/AGC, or is it controlled lighting conditions? You appear to have fixed gain and shutter, so presumably [...]

It is controlled lightning, with a fixed low shutter and gain. I'm trying to achieve stable 120fps for about 1.5min of continuous recording. I'm trying to use libcamera because it is more convenient than diving into V4L2 commands, which I'm not really familiar with as for now, but of course I'll do it if I'll have to. You don't think there's a way to get what I need with libcamera-vid or libcamera-raw? (RAW frames are not really necessary if I could get stable framerate with libcamera-vid)
Thank you for the help.

6by9, 2023-02-15:
I'm just considering unnecessary system load.

AIUI libcamera-raw will be running all frames through the ISP to generate stats that you're then discarding as AE/AGC is locked, AWB is irrelevant for a mono sensor, and you're saving the raws only so lens shading is irrelevant.
Using V4L2 directly removes all that redundant processing.

libcamera-vid has been used to capture 720p118 from a 10bit sensor, so your 640x400@120 should be fine.

You say that there are frame skips/drops. How are you determining that?
I believe the ```--save-pts <filename>``` should work in libcamera-raw, so you could test by sending the output to /dev/null but saving the timestamps to see if there are actual frame drops or it is the saving that is slowing things down.
A similar test should be possible for libcamera-vid.

naushir, 2023-02-23:
Is there anything more needed here?  If not, I'll close this issue down.

closed 2023-02-23 by GuyPerets106