raspberrypi/rpicam-apps #472: Unable to set controls: Input/output error - CSI -> HDMI connection

RagdollCar, 2023-02-15:
I am using IMX477 connected to RPI4. There is also LCD screen. All stuff is as on the photo below:

[image]

I am using CSI to HDMI adapter dedicated for this camera. When trying to launch the application `libcamera-vid` I get an error. Logs are attached below:

```
[0:00:37.133010347] [1064]  WARN IPAManager ipa_manager.cpp:114 Public key not valid
[0:00:37.166577087] [1064]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4171-9b860a66
[0:00:37.284314254] [1065]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media2 and ISP device /dev/media3
[0:00:37.286832013] [1064]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420
[0:00:37.288306772] [1065]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:00:38.131857605] [1065] ERROR V4L2 v4l2_device.cpp:352 'imx477 10-001a': Unable to set controls: Input/output error
```

Does it require any changes in `/boot/config.txt` to enable connection via HDMI?

naushir, 2023-02-15:
Unfortunately you will have to get support directly from Arducam for their products.

closed 2023-02-15 by naushir

RagdollCar, 2023-02-18:
Updated solution: change HDMI cable.