raspberrypi/rpicam-apps #473: libEGL warning: DRI2: failed to authenticate

Petros626, 2023-02-20:
Hello guys,

`OS: Raspberry Pi OS (64-bit)`

I have like so many here also problems with the camera to start:

**libEGL warning: DRI2: failed to authenticate**

Until recently it worked without a problem after a suggested update, but not anymore. I also don't want to use the` --qt-preview` (which makes the camera work) as this is not recommended.

> Using the preview window based on the QT framework, this setting **is not recommended** under normal circumstances, because the preview demo will not use zero-copy buffer sharing and GPU acceleration, which will occupy more resources. The QT preview window supports X forwarding (the default preview program does not).

Kann mir jemand sagen wie ich dieses Update beheben/rückgängig machen kann oder gibt es mittlerweile eine Lösung die Befehle ohne das  `--qt-preview` zu nutzen?

Vielen Dank

naushir, 2023-02-20:
Can you copy/paste the output of the ``camera-bug-report`` tool please?

Petros626, 2023-02-20:
> 

Thank you for answering the same day!

Sure if it's just executed by typing this command, this is my output:

(in order not to overload the page with information, i can also share the file as google drive link - just tell me)

--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 20-02-2023 (19:51:53)
Command: /usr/bin/camera-bug-report

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: d03115
Serial		: 1000000034feab90
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=6d9ac07d-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

** cat /boot/config.txt **
# For more options and information see
[... 893 lines cut ...]
--------------------------------------------------------------------------------
Versions
--------------------------------------------------------------------------------
** uname -a **
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

** cat /etc/os-release **
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Jan  5 2023 10:46:54 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)

** libcamera-hello --version **
libcamera-apps build: 9f08463997b8 26-01-2023 (16:35:25)
libcamera build: v0.0.3+40-9b860a66

naushir, 2023-02-20:
Are you running the board headless or over SSH? If so, you will get this error message as the egl preview window cannot be remoted. In such cases, you will have to use the qt preview window.

Petros626, 2023-02-21:
> Are you running the board headless or over SSH? If so, you will get this error message as the egl preview window cannot be remoted. In such cases, you will have to use the qt preview window.

I'm running it with with a display, keyboard etc. so **no** SSH. I have also read about this, but it does not apply in my case. I only know that I started a suggested update yesterday and after that the camera could no longer be started. 

Do you suggest a new installation of the specific drivers?

naushir, 2023-02-21:
Hmmm, that does seem strange indeed.  I'm not sure, but perhaps the update process failed and you are stuck with an inconsistent set of libraries? You could try doing the following:

```
sudo apt purge libcamera-apps
sudo apt install -y libcamera-apps
```

and see if that fixes the situation.  Alternatively, it would be worth trying a fresh sdcard image (and an update on top of that) if you are able to.

Petros626, 2023-02-21:
> Hmmm, that does seem strange indeed. I'm not sure, but perhaps the update process failed and you are stuck with an inconsistent set of libraries? You could try doing the following:
>
> ```
> sudo apt [...]

I'll try to remove and reinstall are there further drivers or packages, which work with `libcamera-apps` together and should be reinstalled?

You'll get an update soon

naushir, 2023-02-21:
> I'll try to remove and reinstall are there further drivers or packages, which work with `libcamera-apps` together and should be reinstalled?

There are a few dependencies:

```
pi@pi4:~ $ apt-cache show libcamera-apps
Package: libcamera-apps
Version: 0~git20230125+9f08463-1
Architecture: arm64
Maintainer: Serge Schneider <serge@raspberrypi.com>
Installed-Size: 1573
Depends: libavcodec58 (>= 7:4.2), libavdevice58 (>= 7:4.0), libavformat58 (>= 7:4.2), libboost-program-options1.74.0 (>= 1.74.0), libc6 (>= 2.29), libcamera0 (>= 0~git20230124+9b860a66), libdrm2 (>= 2.4.99), libepoxy0 (>= 1.0), libexif12 (>= 0.6.21-1~), libgcc-s1 (>= 3.0), libjpeg62-turbo (>= 1.3.1), libpng16-16 (>= 1.6.2-1), libqt5core5a (>= 5.0.2), libqt5gui5 (>= 5.0.2) | libqt5gui5-gles (>= 5.0.2), libqt5widgets5 (>= 5.0.2), libstdc++6 (>= 7), libswresample3 (>= 7:4.0), libtiff5 (>= 4.0.3), libx11-6
Conflicts: libcamera-apps-lite
Replaces: libcamera-apps-lite
Homepage: https://github.com/raspberrypi/libcamera-apps/
Priority: optional
Section: video
Filename: pool/main/libc/libcamera-apps/libcamera-apps_0~git20230125+9f08463-1_arm64.deb
Size: 299648
SHA256: 5ff802ae4ba0e5a1f8dcff4bad0e39c9bf84ead06a5d8373a02dcf4b6c057f1e
SHA1: 4f7e4ad76325dabc5a526e7bb78c2faeecd6ad99
MD5sum: ff267f9f5fd90a127c76ed6aa6f20ee1
Description: libcamera-apps
 Small suite of libcamera-based apps that aim to copy the functionality of the
 existing "raspicam" apps
Description-md5: eaec0c6bb8e262e76ae4dd4bc00a12bd
```

but to start with, I would just try with purging ``libcamera-apps`` and see what happens.

Petros626, 2023-02-22:
Hello,

In the end, nothing has changed, not even after a new installation. You could of course use a new image, but who can guarantee that the same problem will not occur after an update? Perhaps the update should only be done via the commands sudo apt-get update and sudo apt-get (full-) upgrade.

Is the Raspberry Pi Foundation working on a solution?

naushir, 2023-02-22:
> You could of course use a new image, but who can guarantee that the same problem will not occur after an update

Unfortunately you seem to have hit some unknown problem with what I think is inconstancy between your local installed libraries and the updated libraries.  This is the first report I've seen of such an issue.  I've just tried an update on my existing installation, and with a brand new sdcard image, and both seemed to work as expected.  So I don't suspect anything wrong with the latest build of the libraries.

Really the only suggestion I have is to restart with a fresh sdcard image.  Sorry about that :-(

> Perhaps the update should only be done via the commands `sudo apt-get update` and ` sudo apt-get (full-) upgrade`.

All updates  use the apt mechanism under the hood, even through the gui.

closed 2023-02-22 by Petros626

Petros626, 2023-02-24:
@naushir 

Hey after I reinstalled the Bullseye and just overclock my GPU, CPU and installing my display drivers. I tried to use my camera, but I get still the same error. Some people recomment to switch to Buster 32 OS. I dont know why it worked in the past and then brokes

naushir, 2023-02-24:
> @naushir
> 
> Hey after I reinstalled the Bullseye and just overclock my GPU, CPU and installing my display drivers. I tried to use my camera, but I get still the same error. Some people recomment to switch to Buster 32 OS. I dont know why it worked in the past and then brokes

So this is with a freshly install sdcard image?   This all seems very strange.  A fresh OS image really ought to work.

When you say "installing my display drivers" what steps are you doing for this?  Is this an HDMI display, or some kind of DSI panel?

Petros626, 2023-02-24:
Yes with a freshly installed sd-card hahaha. For met too, I mean I did the following changes:
`#uncomment to overclock the arm. 700 MHz is the default.`
`over_voltage=3`
`arm_freq=1850`
`gpu_freq=550`

It's a DSI Display and **not** plugged in,when I'm using the camera. 
I followed the tutorial from Waveshare (https://www.waveshare.com/wiki/10.1inch_DSI_LCD_(C)) step: **Method 1: Install Manually**. also I installed the step: **Method 1: Graphical Interface Dimming**. 

The only change in my config.txt file after all (display works fine) is:

`# settings for the waveshare display (10.1inch Capacitive Touch Display).`
`ignore_lcd=1`
`dtparam=i2c_vc=on`
`dtparam=i2c_arm=on`
`dtoverlay=WS_xinchDSI_Screen,SCREEN_type=8,I2C_bus=10` --> is this the reason of my problem ?
`dtoverlay=WS_xinchDSI_Touch,I2C_bus=10,invertedx,swappedxy` -> this too ?

naushir, 2023-02-24:
The overclocking params should not be a problem.  Can you disable the touchscreen overlays and try a monitor/TV over HDMI.

[This section](https://www.raspberrypi.com/documentation/computers/camera_software.html#getting-started) in the documentation might also be relevant to your problem.

6by9, 2023-02-24:
> It's a DSI Display and not plugged in,when I'm using the camera.

That'll be the problem.
Device tree has defined the relevant modules for your display, so the system is expecting to find them. If any of those modules fails to probe (eg because they fail to find the relevant piece of hardware), then the entire vc4 DRM module won't bind, and you have no DRM subsystem loaded. No DRM means no route for libcamera to render frames.

You could check for modules that are being waited upon via a combination of ```sudo cat /sys/kernel/debug/devices_deferred``` (should list no devices as having deferred), and ```sudo cat /sys/kernel/debug/device_component/gpu``` (should list about 9 devices, all as "bound").

Petros626, 2023-02-24:
> The overclocking params should not be a problem. Can you disable the touchscreen overlays and try a monitor/TV over HDMI.
> 
> [This section](https://www.raspberrypi.com/documentation/computers/camera_software.html#getting-started) (see the first two _Notes_) in the documentation might also be relevant to your problem.

**I solved it**... it's the two marked lines of the 10.1 inch display in my `config.txt `file like @6by9 mentioned. 

Output 1: `No devices having deferred`

Output 2: 

```
master name                                            status
-------------------------------------------------------------
gpu                                                     bound

device name                                            status
-------------------------------------------------------------
fe600000.firmwarekms                                    bound

```

If I understand it right the display must be connected, so that the camera works fine? 
Is there a way to decouple this from each other, for example, using the camera without the display and using the display without the camera?

6by9, 2023-02-24:
If you're using vc4-fkms-v3d, then you can't be using the waveshare overlays as those rely on vc4-kms-v3d (note fkms vs kms).

For a standard vc4-kms-v3d Pi4 without DSI display I get 
```
pi@raspberrypi:~ $ sudo cat /sys/kernel/debug/device_component/gpu
aggregate_device name                                  status
-------------------------------------------------------------
gpu                                                     bound

device name                                            status
-------------------------------------------------------------
fe400000.hvs                                            bound
fef00700.hdmi                                           bound
fef05700.hdmi                                           bound
fe004000.txp                                            bound
fe206000.pixelvalve                                     bound
fe207000.pixelvalve                                     bound
fe20a000.pixelvalve                                     bound
fe216000.pixelvalve                                     bound
fec12000.pixelvalve                                     bound
```
With your display configured I would expect to find additional entry
```
fe700000.dsi                                            bound
```

Waveshare are being naughty with regard the GPLv2 licence of the Linux kernel and haven't published source for their modules. That means we can't review them to see what is potentially going on or not.
https://github.com/waveshare/Waveshare-DSI-LCD/issues/4

The display and camera should work in any combination, but you can not define hardware as being present when it isn't. You may find using a [gpioN=Y] conditional in config.txt to enable/disable hardware. https://www.raspberrypi.com/documentation/computers/config_txt.html#the-gpio-filter

Petros626, 2023-02-24:
Thank you for the for the detailed answer. 

I think display and camera in combination should allow a preview window without problems, whether with libcamera-apps or the Python library picamera2 or OpenCV. I will test the behavior without HDMI display and if necessary add the post again.

The GPIO filter is only for hdmi not for dsi as I understand.

6by9, 2023-02-24:
> The GPIO filter is only for hdmi not for dsi as I understand.

The GPIO filter allows selection of options based on the state of GPIO lines, eg attach a wire loop from GPIO2 to GND if you want the display enabled.

Are you confusing it with the EDID filter? That is HDMI only.

Petros626, 2023-02-26:
> > The GPIO filter is only for hdmi not for dsi as I understand.
> 
> The GPIO filter allows selection of options based on the state of GPIO lines, eg attach a wire loop from GPIO2 to GND if you want the display enabled.
> 
> Are you confusing it with the EDID filter? That is HDMI only.

The display cannot be controlled via one of the GPIOs on the RPi 4, it is connected directly to the display interface via a DSI ribbon cable. I can't answer the second question because I haven't done it with this filter.

It only works if both are present (display&camera), ` config.txt ` is currently designed for this by commenting out and restarting you can bypass this (using one hardware without the dependency of the other). 

Under [all] the display settings are listed, maybe it is possible to use the camera with HDMI display with **Combining Conditional Filters**.

If I get a working solution I'll ad it here for those, who want to use a Camera with or without a DSI-Display/HDMI-Display