raspberrypi/rpicam-apps #475: Question: why is DRM preview failing if it is not master

pietermarsman, 2023-02-21:
Hi, 

Thanks for the wonderful library! 

We are using libcamera as part of a larger application and want to render the preview on top of other UI elements. But unfortunately this is is not possible using drm because it needs to be the master. I could not find any documentation or git commit message that explains why that is the case. Could you provide a bit more context on why libcamera-apps does not work without being drm master? That would set us off into the right direction? 

[This](https://github.com/raspberrypi/libcamera-apps/pull/14.) is the MR that changed it, so maybe @davidplowman  knows more?

RagdollCar, 2023-02-21:
I have also run into this problem and ended up using terminal mode without Window Manager.

davidplowman, 2023-02-21:
Just to clear up some terminology, libcamera-apps does work without being DRM master, and indeed it has to work like this when X-Windows is running. So far as I understand it, X-Windows insists on being the sole owner of the display, and won't let anyone else use DRM planes. Consequently the only way to render things in this setup is to use various X-Windows frameworks and toolkits.

To use DRM you have to suspend or stop X-Windows entirely. In this environment you can be the DRM master and can draw to the display as you wish.

libcamera-apps aren't really designed to work with GUI elements. Depending on what your use case is you might consider looking at Picamera2 which lets you write Qt applications and use overlays.

pietermarsman, 2023-02-21:
Thanks @davidplowman! So do I understand correctly that the check for DRM master is not because libcamera-apps needs it? But because an additional X-Windows framework or toolkit is needed to make it work without. 

Do you happen to have a recommendation for an X-Windows framework or toolkit that we can use to render the DRM preview over other things? For your information, the app needs to be performant and well integrated in the UI, we decided to move away from QT, and therefore prefer DRM over EGL or Qt. 

Thanks a lot for your kind help! I feel like my questions are awkwardly inaccurate, but you manage to give helpful answers regardless.

6by9, 2023-02-21:
Potentially useful link
https://en.wikipedia.org/wiki/Direct_Rendering_Manager#DRM-Master_and_DRM-Auth
DRM-Auth may provide an option to allow a secondary client to also render planes.

DRM uses explicit plane numbers, so if you have two uncoordinated apps trying to render things, there is nothing stopping them both choosing the same plane and stomping on each other.
The one thing that may trip up DRM-Auth is that I'm not sure you can get vblank callbacks for when a commit has completed. There are certain calls that really are restricted to the master only.

To add regarding when in X, composition is done via GL rather than DRM planes. This is less efficient, but is designed around x86 desktops which have lots of graphics grunt.
With Wayland there is the potential to use multiple DRM planes to compose, but there are sections that are still to be implemented.

davidplowman, 2023-02-21:
I certainly understand the desire to use DRM because it's so much more efficient than going through X-Windows, even with EGL. Nonetheless, the X-Windows solution is pretty good on a Pi 4 - but obviously it all depends what platform and performance targets you want to hit.

naushir, 2023-02-23:
There has been a recent PR merged (https://github.com/raspberrypi/libcamera-apps/pull/471) that significantly improves the CPU performance of the QT preview method.  This might help in your use case here.

closed 2023-03-03 by pietermarsman