raspberrypi/rpicam-apps #478: [BUG]Raspberrypi 4B + Camera Module 3 *** no cameras available ***

MZIchenjl, 2023-02-22:
**Describe the bug**
Raspberrypi Camera Module 3 can't be recognized by my Raspberrypi 4B

```
pi@raspberrypi:~ $ libcamera-hello
Preview window unavailable
[0:00:20.076342801] [684]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
ERROR: *** no cameras available ***
```

I have ran command `sudo apt update && sudo apt upgrade` && `sudo rpi-update`

**Bug report**

```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 22-02-2023 (22:38:47)
Command: /usr/bin/camera-bug-report -t 5 -o bug.txt -c libcamera-still -t 1000 -o test.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: d03115
Serial		: 10000000f587ee9c
Model		: Raspberry Pi 4 Model B Rev 1.5

--------------------------------------------------------------------------------
[... 836 lines cut ...]
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

** vcgencmd version **
Feb 22 2023 10:46:02 
Copyright (c) 2012 Broadcom
version 74a4b109e7f5be465332a1f102649d34f8498d05 (clean) (release) (start)

** libcamera-still --version **
libcamera-apps build: 9f08463997b8 26-01-2023 (16:35:25)
libcamera build: v0.0.3+40-9b860a66

--------------------------------------------------------------------------------
libcamera-still -t 1000 -o test.jpg
--------------------------------------------------------------------------------
Preview window unavailable
[0:06:00.248046125] [793] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:299 [0mlibcamera v0.0.3+40-9b860a66
ERROR: *** no cameras available ***

```

**Additional context**
Add any other context about the problem here.

6by9, 2023-02-22:
Can you post the output from ```sudo i2cdetect -y 10```? You may need to ```sudo apt install i2c-tools``` if i2cdetect isn't already installed, or ```dtparam=i2c_vc=on``` if it reports /dev/i2c-10 as not existing.

Your current logs would imply that the firmware hasn't detected the sensor at all, and therefore nothing has been enabled.

MZIchenjl, 2023-02-23:
@6by9 thanks, i finally found it's the problem of the wire

closed 2023-02-23 by MZIchenjl