raspberrypi/rpicam-apps #480: Algorithm settings

Petros626, 2023-02-27:
Based on the sharpness algorithm, it would be interesting to know how the values are taken over from a command with this param:
`--sharpness, -sh ` #Set image sharpness (-100 - 100)

regarding the sharpness algorithm parameters, which can also be entered/set in a specific `.json ` file (depending on camera model).

`threshold 1.0 `
`strength 1.0` 
`limit 1.0 `

For example --sharpness 75 sets all the params to this value I guess?

naushir, 2023-02-28:
Not exactly, we also factor in if we use binning modes or not.  You can see exactly what the parameter does over [here](https://github.com/raspberrypi/libcamera/blob/9b860a664e9cd7ca4889c6f8d7f0e8d402199de3/src/ipa/raspberrypi/controller/rpi/sharpen.cpp#L64).  ``userStrength_`` is the parameter that gets set with the ``--sharpness`` command line argument.

closed 2023-02-28 by Petros626

naushir, 2023-02-28:
You are right, we do need to update that document with the latest code changes.