raspberrypi/rpicam-apps #481: Framerate specified is deviating

GuyPerets106, 2023-02-28:
When I specify _--framerate X_ there are persistent deviations from the value. 60fps becomes 57, 120fps becomes 115, 180 becomes 172, etc. I use _libcamera-vid_ and _raw_ , and get the same problem with both at any resolution. There's a way to set the framerate more accurately? 
I use _v4l2-ctl --stream-mmap --stream-count=-1 -d /dev/video0 --stream-to=/dev/null_ to check the framerate.

Thank you

6by9, 2023-02-28:
What sensor are you using?

6by9, 2023-02-28:
OV9281 is not a module made by Raspberry Pi. Support is therefore limited and makes assumptions on the module implementation based on those seen in the wild (eg that it uses a 24MHz oscillator).

I've just tested with an Arducam UC-599 OV9281.
Setting /sys/class/video4linux/video0/dev_debug to 3 and then running ```libcamera-hello -t 0 --framerate 60``` gives me the timestamps in the kernel log:
```
[  358.348520] frame_sequence=37
[  358.365132] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=38, id=0, timestamp=358.353797922
[  358.365174] frame_sequence=38
[  358.381802] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=39, id=0, timestamp=358.370464381
[  358.381845] frame_sequence=39
[  358.398468] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=40, id=0, timestamp=358.387131192
[  358.398509] frame_sequence=40
[  358.415132] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=41, id=0, timestamp=358.403797652
[  358.415175] frame_sequence=41
[  358.431798] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=42, id=0, timestamp=358.420464797
[  358.431841] frame_sequence=42
[  358.448473] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=43, id=0, timestamp=358.437131219
[  358.448514] frame_sequence=43
[  358.465131] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=44, id=0, timestamp=358.453797493
[  358.465174] frame_sequence=44
[  358.481798] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=45, id=0, timestamp=358.470463508
``
All fairly accurately 16.6ms.
Ask for 120fps and that gives an accurate 8.33ms.
In both cases using v4l2-ctl after libcamera-hello gives the frame rate requested.

I've also previously tested with Innomaker OV9281 modules and had the expected results.

Check with Waveshare that they are using a the expected oscillator, or you may have a dodgy module.

naushir, 2023-03-06:
Is there anything more to look at here?  If not, I'll close this issue down shortly.

closed 2023-03-06 by GuyPerets106