raspberrypi/rpicam-apps #484: Update broke libcamera-apps ?

Petros626, 2023-03-05:
Hey guys,

It seems that the last update of `python3-libcamera `broke the use of libcamera-apps:

`libcamera-hello: error while loading shared libraries: libcamera.so.0.0.3: cannot open shared object file: No such file or directory`

Just to be sure does anyone else have these problems or it's just me. I wonder about how `libcamera.so.0.0.3` is responsible for this, because with the `find` command I see only `libcamera.so.0.0.4`

naushir, 2023-03-06:
Did you have a local build of libcamera-apps installed which perhaps conflicts with the package install?  What is the output of ``which libcamera-hello`` and ``ldd $(which libcamera-hello)``

Petros626, 2023-03-06:
Yes I build the `libcamera-apps` following this instructions (https://www.raspberrypi.com/documentation/computers/camera_software.html#introducing-the-raspberry-pi-cameras), because in Issue (https://github.com/raspberrypi/libcamera-apps/issues/473) I recognized too late (my fault), that it wasn't the `libcamera-apps`, but rather the my `config.txt` file. I think the package install conflicted with the source build, so what would be the way to prefer (always have the latest version).

The outputs are:
`/usr/local/bin/libcamera-hello`
and
`libcamera.so.0.0.4 => /lib/aarch64-linux-gnu/libcamera.so.0.0.4 (0x0000007fa893a000)`
`libcamera-base.so.0.0.4 => /lib/aarch64-linux-gnu/libcamera-base.so.0.0.4 (0x0000007fa8905000)`

naushir, 2023-03-07:
If you don't always want the bleeding edge software, I would suggest stick with the apt libraries which get periodically updated.

Given your output says that libcamera-hello is linked to libcamera.so.0.0.4, I'm a bit surprised with your original problem of trying to load libcamera.so.0.0.3!  What if you try to explicitly run ``/usr/local/bin/libcamera-hello``?

Petros626, 2023-03-13:
Hey,

I cant tell you what is the output of your suggested command, because I deleted everything and reinstall `libcamera-apps`. The weird thing was only, that the update broke my existing `libcamera-apps` (from source). If in future an update destroy again my installation I'll reopen this issue here.

closed 2023-03-13 by Petros626