raspberrypi/rpicam-apps #486: libcamera-vid writes to disk or ram?

GuyPerets106, 2023-03-10:
Hi , I connected an SSD to my setup (which I tested with `dd` and gives write/read speeds of above 300Mbps).
When I'm using libcamera-vid or libcamera-raw I see that it can't reach the framerate I specify (a few frames per second less) and that it quickly lower the framerate to about half of what I specify , sometimes resulting in frame skips or just time-lapsing (things move in the video faster than in real life). I tested this with Innomaker OV9281 and Waveshare OV9281 on a RPi4B + SSD and also CM4+IO board+SSD. 
Is there a way to write directly to the disk? I don't know if that's what libcamera-vid/raw is doing right now , but this problem keeps happening. 

Example code I wrote:
`libcamera-vid --mode 1280:720:8:P --framerate 120 --width 1280 --height 720 --frames 600 --vflip 1 --hflip 1 -o test.h264`
or 
`libcamera-raw mode 1280:720:8:P --framerate 120 -t 5000 --vflip 1 --hflip 1 -o test.raw`

libcamera-vid has a preview in the command line which prints: (50 last frames)
`#549 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#550 (115.15 fps) exp 8299.00 ag 1.25 dg 1.02
#551 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#552 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#553 (115.13 fps) exp 8299.00 ag 1.25 dg 1.03
#554 (57.60 fps) exp 8299.00 ag 1.25 dg 1.03
#555 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#556 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#557 (115.18 fps) exp 8299.00 ag 1.25 dg 1.03
#558 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#559 (57.59 fps) exp 8299.00 ag 1.25 dg 1.03
#560 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#561 (115.19 fps) exp 8299.00 ag 1.25 dg 1.03
#562 (57.59 fps) exp 8299.00 ag 1.25 dg 1.03
#563 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#564 (115.18 fps) exp 8299.00 ag 1.25 dg 1.03
#565 (57.58 fps) exp 8299.00 ag 1.25 dg 1.03
#566 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#567 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#568 (115.19 fps) exp 8299.00 ag 1.25 dg 1.02
#569 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#570 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#571 (115.14 fps) exp 8299.00 ag 1.25 dg 1.02
#572 (57.60 fps) exp 8299.00 ag 1.25 dg 1.02
#573 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#574 (115.14 fps) exp 8299.00 ag 1.25 dg 1.02
#575 (57.60 fps) exp 8299.00 ag 1.25 dg 1.02
#576 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#577 (115.17 fps) exp 8299.00 ag 1.25 dg 1.02
#578 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#579 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#580 (115.15 fps) exp 8299.00 ag 1.25 dg 1.02
#581 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#582 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#583 (115.19 fps) exp 8299.00 ag 1.25 dg 1.02
#584 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#585 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#586 (115.17 fps) exp 8299.00 ag 1.25 dg 1.02
#587 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#588 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#589 (115.14 fps) exp 8299.00 ag 1.25 dg 1.02
#590 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#591 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#592 (115.13 fps) exp 8299.00 ag 1.25 dg 1.02
#593 (57.60 fps) exp 8299.00 ag 1.25 dg 1.02
#594 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#595 (115.13 fps) exp 8299.00 ag 1.25 dg 1.02
#596 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#597 (57.59 fps) exp 8299.00 ag 1.25 dg 1.02
#598 (57.58 fps) exp 8299.00 ag 1.25 dg 1.02
#599 (115.19 fps) exp 8299.00 ag 1.25 dg 1.02`

Notice I DID try to use `-n` to disable the preview or `--level 4.2` and other methods I saw on the web.

Thanks!

naushir, 2023-03-10:
The system is probably just not keeping up with the requested framerate with your configuration.  I've listed some points about high framerate capture in our documentation [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#high-framerate-capture).  Try some of those steps and see if that helps.

6by9, 2023-03-10:
720p120 is at the top end of level 4.2. The encoder hardware is only specced for level 4.0, so anything beyond is not guaranteed. (1080p50 can normally be achieved. 1080p60 is a push).

GuyPerets106, 2023-03-10:
> The system is probably just not keeping up with the requested framerate with your configuration. I've listed some points about high framerate capture in our documentation [...]

What do you mean by 'keeping up'? Can you elaborate on how libcamera-vid works in terms of memory mapping? libcamera-raw works differently? 
Thanks for the link , but I already tried the methods there.

> 720p120 is at the top end of level 4.2. The encoder hardware is only specced for level 4.0, so anything beyond is not guaranteed. (1080p50 can normally be achieved. 1080p60 is a push).

Innomaker has a driver of their own with specific tools (open source on their github) that writes raw video at 253fps 1280x800 resolution , and with an SSD , I never encountered a problem of slowing down. Can you elaborate on how libcamera-vid works compared to libcamera-raw , in terms of memory mapping? SSD is not something that could help? 
By the way , OV9281 sensor can't reach 1080p any way , max is 1280x800 (1MP sensor). 1280x720 is fine for me in terms of resolution , but framerate is something I can't compromise - libcamera tools can't guarantee high stable framerate?

Thanks!

naushir, 2023-03-10:
> What do you mean by 'keeping up'? Can you elaborate on how libcamera-vid works in terms of memory mapping? libcamera-raw works differently? Thanks for the link , but I already tried the methods there.

What I mean is that the entire camera system (sensor + CSI2 + ISP + Video encoder + any software overhead) may be incapable of achieving stable 720p120fps video encode.  As @6by9 mentioned, the encoder hardware is only specced for level 4, so anything beyond is not guaranteed.  Having said that, with the steps outlined in the document I linked above, I have managed to achieve 720p encode at around 115fps with the Raspberry Pi HQ Camera (imx477).  I noticed that your command line does not disable preview - this is quite costly.  Try adding the ``-n`` command line option to disable preview, and you ought to get better results.

``libcamera-vid`` sends the framebuffer through the H.264 video encoder, and saves the encoded stream.  ``libcamera-raw`` directly write the un-encoded framebuffers to file.  When dealing with high framerates, you are much more likely to get closer to the target with ``libcamera-vid``.

> Innomaker has a driver of their own with specific tools (open source on their github) that writes raw video at 253fps 1280x800 resolution , and with an SSD , I never encountered a problem of slowing [...]

Chances are their application is only saving the RAW/Bayer data directly to file, likely without libcamera either.  This is much simpler on resources.  If you need YUV/RGB frame from that, you will have to do a software debayer and other bits of processing to get usable images out.

6by9, 2023-03-10:
> If you need YUV/RGB frame from that, you will have to do a software debayer and other bits of processing to get usable images out.

OV9281 is a mono sensor, so largely it'll be only basic AE / AGC and lens shading that is needed.

naushir, 2023-03-16:
Anything more to look at for this issue?  If not, I'll close it down shortly.

naushir, 2023-03-16:
Resolving now.

closed 2023-03-16 by naushir