raspberrypi/rpicam-apps #487: Compatibility with OpenCV under Bullseye OS

Petros626, 2023-03-13:
Hey,

Currently _libcamera_ is not supported by _OpenCV_, you have to use the old` legacy camera stack` to work with _OpenCV_. 

There are no resources available from _OpenCV_ for this (https://github.com/opencv/opencv/issues/23328), so the question would be why the _Raspberry Pi Foundation_ has developed an incompatible camera support, or whether this will be adapted in the future so that it can be used under _OpenCV_ again.

This can also be explained in a comparison of old and new camera support.

naushir, 2023-03-13:
> Currently _libcamera_ is not supported by _OpenCV_, you have to use the old` legacy camera stack` to work with _OpenCV_.
> 
> There are no resources available from _OpenCV_ for this ([opencv/opencv#23328](https://github.com/opencv/opencv/issues/23328)), so the question would be why the _Raspberry Pi Foundation_ has developed an incompatible camera support, or whether this will be adapted in the future so that it can be used under _OpenCV_ again.

We have moved our camera stack to libcamera for many reasons.  Some of them are outlined [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#raspicam-applications).  You can still integrate libcamera-apps with OpenCV, we documentation describing this [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#post-processing-with-opencv).  What is missing is native libcamera support for OpenCV.  This is up to the OpenCV developers to add when they are able to.

Petros626, 2023-03-13:
Thank you for clarify some things. For example the access to a camera with ``cv.VideoCapture()` is with _libcamera_ not possible. Many users mentioned that in different forums.

naushir, 2023-03-13:
Unfortunately, that is going to be up to the OpenCV developers to integrate libcamera into their stack.

6by9, 2023-03-13:
The original OpenCV issue requesting native libcamera support is https://github.com/opencv/opencv/issues/21653