raspberrypi/rpicam-apps #488: libcamera-still in signal mode exits efter 1st SIGUSR1 signal

mnellemann, 2023-03-13:
Hi,

When running libcamera-still with ```--signal --timeout=0``` the process stops after receiving first SIGUSR1 (```kill -s SIGUSR1 pid```).

```shell
# libcamera-still --nopreview --immediate --flush --output=/dev/shm/capture.jpg --signal --timeout=0
[0:31:08.794676681] [3908]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:31:08.858832410] [3909]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[0:31:08.860913139] [3908]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUV420 (1) 2592x1944-SGBRG10_CSI2P
[0:31:08.861434181] [3909]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
Received signal 10
Still capture image received
#
```
The image is saved as expected, but process exits.

This is on a Raspberry Pi Zero W2 running latest DietPi (Debian 11.6).

```shell
# uname -a
Linux camera-00000000f49cc010 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
# libcamera-still --version
libcamera-apps build: 54a781dffdd1 01-03-2023 (17:53:36)
libcamera build: v0.0.4+22-923f5d70
```

Not sure if I'm doing this wrong or if this could have been fixed in a newer version?

naushir, 2023-03-14:
This is the intended behaviour of ``libcamera-still``.  The signal will result in the still image capture being taken, after which the application will terminate.

mnellemann, 2023-03-14:
Thanks - this was not clear to me. If I recall correctly it was possible with the old *raspistill*.

Are there any recommendations to get the behavior I am looking for - capturing images with minimum delay ?

naushir, 2023-03-15:
Actually, looking at the code a bit more, if you don't use ``--immediate`` it might do exactly what you want?  See [here](https://github.com/raspberrypi/libcamera-apps/blob/21157c5001a0a090cbd69c9e04c3a9dcfdfa8617/apps/libcamera_still.cpp#L308).

mnellemann, 2023-03-15:
You are correct, many thanks.

closed 2023-03-15 by mnellemann