raspberrypi/rpicam-apps #490: `--codec libav --libav-format mpegts` reports B-frames and gop size argument errors

dtbaker, 2023-03-14:
## Example:

- **Failed to set number of B-frames: Invalid argument**
- **Failed to set gop size: Invalid argument**

```
$ libcamera-vid -n -t 0 --framerate 15 --width 800 --height 600 --bitrate 1000000 --codec libav --libav-format mpegts --libav-audio 0 -o /tmp/test.mpeg

[h264_v4l2m2m @ 0x5582209230]  <<< v4l2_encode_init: fmt=181/0
[h264_v4l2m2m @ 0x5582209230] Using device /dev/video11
[h264_v4l2m2m @ 0x5582209230] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x5582209230] requesting formats: output=YU12 capture=H264
[h264_v4l2m2m @ 0x5582209230] Failed to set number of B-frames: Invalid argument
[h264_v4l2m2m @ 0x5582209230] Failed to set gop size: Invalid argument
Output #0, mpegts, to '/tmp/test-mjpegts2.ts':
    Stream #0:0: Video: h264 (h264_v4l2m2m), drm_prime(tv, smpte170m/smpte170m/bt709), 800x600, q=-1--1, 1000 kb/s, 15 fps, 15 tbr, 1000k tbn

```

I tried the `--intra` and `--inline` arguments just for kicks but the same error appeared.

The `ffmpeg` command I'm trying to convert to `libcamera-vid` looks a bit like this: `ffmpeg -i /dev/video0 -f mpegts -codec:v mpeg1video -s 640x360 -b:v 700k -r 25 -bf 0  output.ts`

6by9, 2023-03-14:
Partial duplicate of https://forums.raspberrypi.com/viewtopic.php?t=348848 - please don't duplicate support requests between forums and Github.

Neither are significant issues - FFmpeg is just being overly shouty.

B frames aren't supported, therefore trying to set the property is a client error but largely irrelevant. Even crazier as it's trying to set the value to 0 as FFmpeg's driver acknowledges that it doesn't support B-frames either https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/v4l2_m2m_enc.c#L153

My original reading of the [spec](https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/ext-ctrls-codec.html) was that V4L2_CID_MPEG_VIDEO_H264_I_PERIOD replaced V4L2_CID_MPEG_VIDEO_GOP_SIZE for H264.
It looks like the docs have been updated. As we don't produce I-frames that aren't IDR, the encoder potentially should be implementing GOP_SIZE.

naushir, 2023-03-14:
As @6by9 said, you can safely ignore these warnings.  I'll look to reduce the verbosity of libav so they disapper.

closed 2023-03-14 by naushir