raspberrypi/rpicam-apps #491: Unsupported bayer format when capturing raw dng image from monochrome sensor

tewarid, 2023-03-14:
I am trying to capture raw dng still image using `libcamera-still --raw -o test.jpg` from a monochrome sensor but am getting `ERROR: *** unsupported Bayer format ***` towards the end. jpg still image looks all right.

The sensor format logged by libcamera is `Selected sensor format: 1280x720-Y10_1X10 - Selected unicam format: 1280x720-Y10P`. Sensor driver sets `MEDIA_BUS_FMT_Y10_1X10`.

libcamera-apps version...

```log
$ apt info libcamera-apps
Package: libcamera-apps
Version: 0~git20230301+54a781d-1
```

6by9, 2023-03-14:
https://github.com/raspberrypi/libcamera-apps/issues/424#issuecomment-1431110067

Raspberry Pi do not sell any mono sensors, therefore mono support is fairly well down the priority list.
The code is all open source, so there's nothing stopping anyone else doing the relevant research and implementing it.

closed 2023-03-14 by naushir

tewarid, 2023-03-14:
> Raspberry Pi do not sell any mono sensors, therefore mono support is fairly well down the priority list.

I can understand this from the perspective of libcamera-apps, but I forgot to mention that I am trying to use camera tuning tool in libcamera and need raw dng image to use with it. I would prefer to spend time tuning instead of coding, but I agree that it all being open source goes a long way. Thanks!

6by9, 2023-03-14:
I wouldn't like to think what the Tuning Tool does with mono data.

AWB obviously needs to be told to do nothing.
AE is largely independent of colour format.
Lens shading will still have an effect, but all 4 tables need to be set the same.
Black level too needs to be set the same for all 4 channels.
Most of the other blocks are after demosaic, so it makes little difference if working on a mono or colour sensor.

As a first level hack, creating a colour DNG would probably work with the tuning tool.