raspberrypi/rpicam-apps #492: Does libcamera-raw use ISP?

JustQJ, 2023-03-15:
Hi, I want to know if libcamera-raw uses ISP when generating raw bayer image. What more,  is it possible for me to use the ISP to process   the raw bayer image generated by libcamera-raw and obtain the corresponding rgb image?
Thanks in advance.

JustQJ, 2023-03-16:
No, I want to kown if I can send bayer images that do not directly come  from sensor but come from user application to the ISP for reprocessing.

naushir, 2023-03-16:
At this point , no you cannot do that.  The libcamera API does not yet have support for RAW framebuffer reprocessing through the ISP.  However, this is a feature we are keen to see implemented, so hopefully it will happen soon.