raspberrypi/rpicam-apps #494: Dual cameras recording with SSD

GuyPerets106, 2023-03-26:
Hi, I'm using an 8GB CM4 with IO board which has 2 CSI connectors. I connected 2 Waveshares OV9281 to it and I'm trying to record 1280x720 8bit video at 120fps from both cameras in parallel. An SSD is connected to the IO board which gives me a write speed of around 380MB/sec.
When I'm using
`libcamera-vid --level 4.2 -o testR.h264 --mode 1280:720:8:P --width 1280 --height 720 --framerate 120 --frames 1200 -n --denoise off --camera 1` 
`libcamera-vid --level 4.2 -o testL.h264 --mode 1280:720:8:P --width 1280 --height 720 --framerate 120 --frames 1200 -n --denoise off --camera 0` 
at the same time (from 2 terminals as for now), I get a 10 seconds video from each, which actually took 20sec to record, and the video is at high speed (like a timelapse video), and this is not what I intended. 
Am I doing something wrong? Is there a quick fix for this problem?
Thanks!

6by9, 2023-03-26:
https://datasheets.raspberrypi.com/rpi4/raspberry-pi-4-product-brief.pdf
> Multimedia: H.265 (4Kp60 decode);
> H.264 (1080p60 decode, 1080p30 encode);

1080p30 being H264 level 4.0.
https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels will tell you that limit is 245,760 macroblocks (16x16) per second.

Under many circumstances you can push it up to 1080p60 for level 4.2. Level 4.2 is 522,240 macroblocks per second.

1280x720@120 would be 432000 macroblocks/sec, so above level 4.0, but not needing the full capabilities of level 4.2.

With two simultaneous encodes you're looking at 864000 macroblocks/sec. 
So you're trying to push the encoder at over 3.5x the rated spec. The hardware can't keep up with that and will result in dropped frames.

GuyPerets106, 2023-03-27:
> https://datasheets.raspberrypi.com/rpi4/raspberry-pi-4-product-brief.pdf
>
>
> 1080p30 being H264 level 4.0. https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels will tell you that limit is 245,760 [...]

I managed to record 1280x800 253fps 8bit video with Innomaker OV9281-V2 , using their driver and an SSD for fast write speeds. Of course it's a different situation because that was RAW video , but the thing you're describing sounds like it's related to CM4 hardware. So if that problem is related to hardware encoding using libcamera-vid , does libcamera-raw can reach the same results as Innomaker's driver , without skipping frames?
Their git: https://github.com/INNO-MAKER/CAM-OV9281RAW-V2

6by9, 2023-03-27:
> I managed to record 1280x800 253fps 8bit video with Innomaker OV9281-V2 , using their driver and an SSD for fast write speeds

Really? That exceeds the spec that Omnivision publish for the sensor. They state 1280x800 @ 120fps, although that is 10bit. I don't see dropping to 8 bit over doubling the frame rate (it normally adds 25%.
640x400 @ ~250fps is plausible. Commented [last week](https://github.com/raspberrypi/libcamera/issues/52#issuecomment-1477680914) that 640x400 in 8bit mode can in theory do 320fps. 10bit should do 260fps.

> So if that problem is related to hardware encoding using libcamera-vid , does libcamera-raw can reach the same results as Innomaker's driver , without skipping frames?

It's a limitation of the H264 codec that you're seeing.
Yes libcamera-raw should be able to go faster. There's still a limit to the speed that the ISP can process the frames for statistics purposes, but IIRC that should be up around 400MPix/s for Pi4/CM4.

> Their git: https://github.com/INNO-MAKER/CAM-OV9281RAW-V2

Another GPL licence violation by them - nice. They just don't care as it was raised over 2 years ago for one of their other sensors - https://gitee.com/inno-maker/cam-mipi327raw/issues/I22HTM

GuyPerets106, 2023-03-27:
> Really? That exceeds the spec that Omnivision publish for the sensor. They state 1280x800 @ 120fps, although that is 10bit. I don't see dropping to 8 bit over doubling the frame rate (it normally [...]

Sorry , I meant mode 1 (which gets 144fps at 1280x800 resolution 8bit). 

> It's a limitation of the H264 codec that you're seeing. Yes libcamera-raw should be able to go faster. There's still a limit to the speed that the ISP can process the frames for statistics purposes, [...]
So H264 is the best codec to use in my situation? MJPEG , yuv420 or libav could perform better?

And if libcamera-raw could potentially record what I'm trying to see , how can I know the pixel format of IMX708? (Module 3)
I assume it's SBGGR10 from what I'm seeing in `Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-BG10` but FFmpeg doesn't have this pixel format for conversion , so maybe I'm wrong?

GuyPerets106, 2023-03-29:
By the way , another problem I encounter is 5% decrease in specified framerate using libcamera-vid , at all resolutions and framerates: 
640x400 240fps turns to 229fps
640x400 180fps turns to 172fps
640x400 120fps turns to 114fps
1280x720 120fps turns to 114fps
1280x720 100fps turns to 95fps

I could specify the framerate I want and multiply by 1.05 , but I'm not sure if it's a bug or a problem with my cameras.
I noticed it while recording the stopwatch on my phone (desired 90 seconds becomes 94seconds of "real-life" recording) , and then used 
`v4l2-ctl --stream-mmap --stream-count=-1 -d /dev/video0 --stream-to=/dev/null`
to check and got these numbers I wrote above. 
the command I write in the terminal is
`libcamera-vid --mode width:height:8:P --width WIDTH --height HEIGHT --framerate FPS --frames FPS*90 -n --denoise off --level 4.2 --shutter S --gain G -o file.h264`

GuyPerets106, 2023-04-04:
> Really? That exceeds the spec that Omnivision publish for the sensor. They state 1280x800 @ 120fps, although that is 10bit. I don't see dropping to 8 bit over doubling the frame rate (it normally [...]

By the way , you once told me about the Waveshare board oscillator which has a 12MHz written on it , like it shows on the picture on their website - https://www.waveshare.com/OV9281-120-Camera.htm 
But I checked on the back of the board I actually got , and it shows 24MHz. 
So I'm not sure what is right about it , and that was the only reason I could think about.

6by9, 2023-04-04:
IMX708 is a raw Bayer sensor, hence producing SBGGR10. FFmpeg doesn't generally handle that as it needs significant sensor specific processing to make it useful, and that is what libcamera is all about.

yuv420 for uncompressed frames is largely your only option for very high framerates, but be aware that you're then handling lots of data.

naushir, 2023-04-14:
Anything further to look at here?  If not, I'll close this down shortly.

GuyPerets106, 2023-04-16:
> Anything further to look at here? If not, I'll close this down shortly.

I'm still trying to understand if there's a limitation to memory writing with libcamera-raw and SSD attached. Theoretically, I should have no problem writing raw data at high resolution and high framerate , am I right? You know why I'm still getting frame drops if the data don't go through the H264 encoder?

Schaggo, 2023-04-16:
This maybe anecdotal and might actually have been solved in the current version. I’m running a custom libcamera app designed to capture a defined number of precisely timed frames at max res/fps.

One bottle neck I encountered was memory management. I don’t know the exact number anymore, but it was something like after 10 frames RAW the 11th would be captured with a significant delay.
The solution was to implement allocate memory ahead of time, so that the program allocates buffers for the specified amount of frames before filling them. 

This will be only useful for a couple of frames limited by the system’s RAM, but maybe a worker that writes the buffer array to disk could improve usability of this concept.

naushir, 2023-04-17:
> I'm still trying to understand if there's a limitation to memory writing with libcamera-raw and SSD attached. Theoretically, I should have no problem writing raw data at high resolution and high [...]

``libcamera-raw`` still processes frames through the ISP (since it needs statistics for AE/AWB correction).  As such, you will still need ``--denoise off`` with ``libcamera-raw``.  Can you provide the exact command line you are using with libcamera-raw?

As @Schaggo mentioned, there may be an issue with lack of buffering and slower IO operations cause some frame drops.  You can try increasing the number of buffers used by the ``--buffer-count`` argument.  The default is 6 buffers, so perhaps try 10 and see if that helps?

closed 2023-06-05 by GuyPerets106