raspberrypi/rpicam-apps #495: No frame data was collected.

wangjunjie1996, 2023-03-27:
**Describe the bug**
When using "Libcamera-Still-R-O test.jpg-Width 4000-Height 2000-Gain1-AWB Gain1,1-Shutter 3000000-Immediate", the data is available for the first time, and the picture is normal for 1.7M, but there is no data behind it, and the picture is only 127kb in black.

**Bug report**
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/11076132/bug.txt)

**Additional context**
I view the information by adding my own log and compiling and installing locally. The following screenshot shows the information.  
[image]  
[image]  

Normal should be as follows  

[image]

And the buf.sequence in "V4L2VideoDevice::dequeueBuffer" is 0,  and it is normal only if it is not 0.

naushir, 2023-03-27:
Would you be able to attach the file created when you run the ``camera-bug-report`` tool to this issue please?

naushir, 2023-03-27:
Given you are using an Arducam camera (and associated closed source libraries), we cannot provide support for this.  Please seek out help in the Arducam forums.

closed 2023-03-27 by naushir