raspberrypi/rpicam-apps #498: [BUG] Raspberry Pi v3 Wide very unstable when Raspberry Pi 7" DSI display is connected.

meantux, 2023-03-30:
**Describe the bug**
When trying to run both the Raspberry Pi 7" DSI display with a Raspberry Pi v3 Wide camera together, the camera fails after a few shots (sometimes after 2 shots, sometimes after 20 shots, mostly within 6 shots)

The screen works fine.

The camera works fine and never fails when the screen is not connected
When both are connected that where the camera is unstable.
Once the error occurs, I absolutely need to reboot to recuperate the use of the camera (and sometimes I have to cut off power):

ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video1[14:cap]: Failed to start streaming: Input/output error
ERROR: *** failed to start camera ***

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.
sudo camera-bug-report -t 1000 -o cbr.txt -c /usr/bin/libcamera-still

[cbr.txt](https://github.com/raspberrypi/libcamera-apps/files/11114432/cbr.txt)

naushir, 2023-03-31:
Can you try disabling the touchscreen and see if that improves things?  To do this, add `` disable_touchscreen=1`` to the bottom of you ``/boot/config.txt`` file and reboot.

6by9, 2023-03-31:
```disable_touchscreen=1``` is for the legacy display stack only.
Change ```display_auto_detect=1``` to ```display_auto_detect=0```, and add ```dtoverlay=vc4-kms-dsi-7inch,disable_touch```

meantux, 2023-03-31:
I got an update on my issue this morning, it was a defective RPi, I swapped it (keeping the same microSD card and soft) and no issue is occurring, now both camera and display are working together. I know it's not just a cable issue though because I did swap a lot of cables, long and shorts, yesterday.

The display doesn't report I/O errors anymore.
The camera is stable.

Thank you for your time.

closed 2023-03-31 by meantux