raspberrypi/rpicam-apps #499: libcamera-still kernel warning on Raspbery Pi ZeroW

wcbonner, 2023-03-31:
I'm getting messages in syslog every time I take a picture. Does this need to be reported here, or with some other repository related to either the camera or kernel?

Here's the basics of my software:
```
wim@WimPiZeroW-Wim:~ $ dpkg -l | grep libcamera
ii  libcamera-apps-lite                    0~git20230301+54a781d-1          armhf        libcamera-apps-lite
ii  libcamera-dev:armhf                    0~git20230302+923f5d70-1         armhf        complex camera support library (development files)
ii  libcamera0:armhf                       0~git20230302+923f5d70-1         armhf        complex camera support library
wim@WimPiZeroW-Wim:~ $ uname -a
Linux WimPiZeroW-Wim 6.1.19+ #1637 Tue Mar 14 11:01:56 GMT 2023 armv6l GNU/Linux
wim@WimPiZeroW-Wim:~ $ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx708_wide_noir [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SBGGR10_CSI2P' : 1536x864 [30.00 fps - (0, 0)/0x0 crop]
                             2304x1296 [30.00 fps - (0, 0)/0x0 crop]
                             4608x2592 [30.00 fps - (0, 0)/0x0 crop]
```
I'm running the command: `libcamera-still --nopreview --thumb none --timeout 49500000 --timelapse 60000 --output /media/WimPiZeroW-Wim/DCIM/20230331/0331%04d.jpg --framestart 1 --verbose 0 --exif 'IFD0.ImageDescription=WimsConstructionCam 1.20230329-2 Built Mar 29 2023 at 15:18:09' --exif 'IFD0.Artist=WimPiZeroW-Wim' --lens-position 0.0`

And the section of /var/log/syslog:
```
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.407508] ------------[ cut here ]------------
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.407545] WARNING: CPU: 0 PID: 21878 at drivers/media/mc/mc-entity.c:873 media_pipeline_stop+0x28/0x34 [mc]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.407753] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep imx708 dw9807_vcm rndis_wlan hci_uart btbcm rndis_host bluetooth cdc_ether brcmfmac ecdh_generic ecc libaes vc4 8021q garp stp llc brcmutil sha256_generic libsha256 cfg80211 snd_soc_hdmi_codec drm_display_helper option cec usb_wwan drm_dma_helper usbserial drm_kms_helper snd_soc_core bcm2835_unicam v4l2_dv_timings i2c_mux_pinctrl i2c_mux bcm2835_codec(C) v4l2_fwnode v4l2_async snd_compress snd_pcm_dmaengine syscopyarea v4l2_mem2mem bcm2835_isp(C) bcm2835_v4l2(C) sysfillrect sysimgblt fb_sys_fops bcm2835_mmal_vchiq(C) raspberrypi_hwmon rfkill videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) snd_pcm videodev snd_timer i2c_bcm2835 snd vc_sm_cma(C) mc uio_pdrv_genirq uio fixed drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408299] CPU: 0 PID: 21878 Comm: libcamera-still Tainted: G        WC         6.1.19+ #1637
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408322] Hardware name: BCM2835
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408339]  unwind_backtrace from show_stack+0x18/0x1c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408379]  show_stack from dump_stack_lvl+0x34/0x58
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408412]  dump_stack_lvl from __warn+0x7c/0xf8
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408446]  __warn from warn_slowpath_fmt+0x60/0xcc
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408483]  warn_slowpath_fmt from media_pipeline_stop+0x28/0x34 [mc]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408641]  media_pipeline_stop [mc] from unicam_stop_streaming+0x88/0x168 [bcm2835_unicam]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.408832]  unicam_stop_streaming [bcm2835_unicam] from __vb2_queue_cancel+0x30/0x284 [videobuf2_common]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.409044]  __vb2_queue_cancel [videobuf2_common] from vb2_core_streamoff+0x20/0xb8 [videobuf2_common]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.409230]  vb2_core_streamoff [videobuf2_common] from __video_do_ioctl+0x224/0x458 [videodev]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.409962]  __video_do_ioctl [videodev] from video_usercopy+0x118/0x574 [videodev]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.410788]  video_usercopy [videodev] from sys_ioctl+0x10c/0x90c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.411240]  sys_ioctl from ret_fast_syscall+0x0/0x1c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.411267] Exception stack(0xdc8c9fa8 to 0xdc8c9ff0)
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.411288] 9fa0:                   b50162b0 b500b5dc 0000000d 40045613 b5012f08 00000000
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.411306] 9fc0: b50162b0 b500b5dc b500b5f0 00000036 01c99420 00000000 b6ef0134 01c993c0
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.411320] 9fe0: b6edcd44 b59544cc b6e5a940 b6aded4c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844203.411332] ---[ end trace 0000000000000000 ]---
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.343481] ------------[ cut here ]------------
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.343520] WARNING: CPU: 0 PID: 21878 at drivers/media/mc/mc-entity.c:873 media_pipeline_stop+0x28/0x34 [mc]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.343781] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep imx708 dw9807_vcm rndis_wlan hci_uart btbcm rndis_host bluetooth cdc_ether brcmfmac ecdh_generic ecc libaes vc4 8021q garp stp llc brcmutil sha256_generic libsha256 cfg80211 snd_soc_hdmi_codec drm_display_helper option cec usb_wwan drm_dma_helper usbserial drm_kms_helper snd_soc_core bcm2835_unicam v4l2_dv_timings i2c_mux_pinctrl i2c_mux bcm2835_codec(C) v4l2_fwnode v4l2_async snd_compress snd_pcm_dmaengine syscopyarea v4l2_mem2mem bcm2835_isp(C) bcm2835_v4l2(C) sysfillrect sysimgblt fb_sys_fops bcm2835_mmal_vchiq(C) raspberrypi_hwmon rfkill videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(C) snd_pcm videodev snd_timer i2c_bcm2835 snd vc_sm_cma(C) mc uio_pdrv_genirq uio fixed drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344406] CPU: 0 PID: 21878 Comm: libcamera-still Tainted: G        WC         6.1.19+ #1637
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344433] Hardware name: BCM2835
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344453]  unwind_backtrace from show_stack+0x18/0x1c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344502]  show_stack from dump_stack_lvl+0x34/0x58
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344544]  dump_stack_lvl from __warn+0x7c/0xf8
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344587]  __warn from warn_slowpath_fmt+0x60/0xcc
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344633]  warn_slowpath_fmt from media_pipeline_stop+0x28/0x34 [mc]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.344828]  media_pipeline_stop [mc] from unicam_stop_streaming+0x88/0x168 [bcm2835_unicam]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.345074]  unicam_stop_streaming [bcm2835_unicam] from __vb2_queue_cancel+0x30/0x284 [videobuf2_common]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.345342]  __vb2_queue_cancel [videobuf2_common] from vb2_core_streamoff+0x20/0xb8 [videobuf2_common]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.345592]  vb2_core_streamoff [videobuf2_common] from __video_do_ioctl+0x224/0x458 [videodev]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.346489]  __video_do_ioctl [videodev] from video_usercopy+0x118/0x574 [videodev]
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.347552]  video_usercopy [videodev] from sys_ioctl+0x10c/0x90c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.348129]  sys_ioctl from ret_fast_syscall+0x0/0x1c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.348168] Exception stack(0xdc8c9fa8 to 0xdc8c9ff0)
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.348194] 9fa0:                   b50162b0 b500b5dc 0000000d 40045613 b5012f08 00000000
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.348218] 9fc0: b50162b0 b500b5dc b500b5f0 00000036 01c99420 00000000 b6ef0134 01c993c0
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.348237] 9fe0: b6edcd44 b59544cc b6e5a940 b6aded4c
Mar 31 10:08:38 WimPiZeroW-Wim kernel: [844204.348390] ---[ end trace 0000000000000000 ]---
```

6by9, 2023-03-31:
See https://github.com/raspberrypi/linux/pull/5409

naushir, 2023-04-03:
As mentioned, this should be resolved with the above commit merged into the kernel.

closed 2023-04-03 by naushir