raspberrypi/rpicam-apps #501: How to get accurate timestamps for frames

ptoews, 2023-04-06:
I want to get accurate absolute timestamps for each recorded frame, ideally with an accuracy of <10ms. 
My first approach consisted of simply setting the output filename to system time right before calling `libcamera-vid`, assuming this is close to the first frame, and the using `--save-pts` to get the offsets for the remaining frames.
However, because of the initialization happening, the first frame is recorded much later after libcamera-vid is started, in my case 8-9 seconds (see #406). 
It gets more accurate using the creation time of the output file, however since I pipe the video through ffmpeg for encoding there might be additional time offsets.

I'm not sure yet if I can get accurate timestamps from ffmpeg.
Is there an easy way to get accurate frame timestamps from libcamera-vid itself?

naushir, 2023-04-08:
The frame timestamps are generated in the kernel at the time of frame acquisition - as such they are indeed accurate to a few micro-seconds of interrupt latency.  Are you looking to output these timestamps relative to the "wall clock"?   If so, this not directly possible with ``libcamera-vid`` (as we have no need for this).  But it should be relatively simple.

You would need to sample the wall clock time on the first frame out of the system, then you have your correspondence between kernel clock and wall clock to do the conversion.  Note that this correspondence will include the libcamera pipeline latencies as well, but not sure if this is a problem for your application.

naushir, 2023-04-14:
Anything further to look at here?  If not, I'll close this down shortly.

ptoews, 2023-04-14:
I'm still trying to get absolute timestamp generation by ffmpeg working, but if it's the only way there is nothing more to discuss I think. Thank you either way.

closed 2023-04-14 by naushir

catubc, 2024-10-03:
hey @naushir , has this issue been resolved?
Indeed, there are some of us trying to use RPi cams to get to ~1ms realtime precision and it doensn't work so well without knowing the wall clock of every video frame.  Taking stills doesn't seem to solve the problem as it disrupts the h.264 compression.

Can you point me to where I can hack this in the libcamera code?   Happy to also contribute to this functionality.
Thanks so much!
catubc