raspberrypi/rpicam-apps #502: [BUG]Using libcamera-vid with '--initial pause' results in corrupted h264 file

ABelliqueux, 2023-04-16:
**Describe the bug**
When using `libcamera-vid` with the `--initial` flag set to 'pause', the resulting h264 file is corrupted

Steps to reproduce :

1. Launch `libcamera-vid -o test.h264 -t 0 -k --initial 'pause'`
2. Hit 'Return' to start recording
3. Wait a few seconds
4. Hit 'Return' to stop recording
5. Hit 'Ctrl+C' to stop libcamera-vid
6. Check 'test.h264' with `ffprobe test.h264`

Results:
```
ffprobe version 4.3.5-0+deb11u1+rpt3 Copyright (c) 2007-2022 the FFmpeg developers
  built with gcc 10 (Raspbian 10.2.1-6+rpi1)
  configuration: [snip]
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[h264 @ 0x16cc540] Format h264 detected only with low score of 1, misdetection possible!
[h264 @ 0x16cd810] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x16cd810] decode_slice_header error
[h264 @ 0x16cd810] no frame!

[snip]

[h264 @ 0x16cd810] decode_slice_header error
[h264 @ 0x16cd810] no frame!
[h264 @ 0x16cd810] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x16cd810] decode_slice_header error
[h264 @ 0x16cd810] no frame!
[h264 @ 0x16cd810] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x16cd810] decode_slice_header error
[h264 @ 0x16cd810] no frame!
[h264 @ 0x16cc540] decoding for stream 0 failed
[h264 @ 0x16cc540] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size
Consider increasing the value for the 'analyzeduration' and 'probesize' options
Input #0, h264, from 'test.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264, none, 25.33 fps, 25 tbr, 1200k tbn, 50 tbc

```
6. Trying the same without the '--initial' flag, i.e; `libcamera-vid -o test.h264 -t 0 -k` results in a perfectly fine file :  
```
$ ffprobe test.h264 
ffprobe version 4.3.5-0+deb11u1+rpt3 Copyright (c) 2007-2022 the FFmpeg developers

[snip]

Input #0, h264, from 'test.h264':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p(progressive), 640x480, 30 fps, 30 tbr, 1200k tbn, 60 tbc
```
**Bug report**

Using `camera-bug-report -t 5 -o bug.txt -c "libcamera-vid -o test.h264 -t 0 -k --initial pause"` and hitting 'Return' once to start recording: 

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/11241964/bug.txt)

naushir, 2023-04-17:
You will need to use ``--inline`` to provide inline headers in the H.264 stream for your use case.  The following command should give you a valid encoded file:
```
libcamera-vid -o test.h264 -t 0 -k --initial 'pause' --inline
```

ABelliqueux, 2023-04-17:
@naushir : Thanks, this indeed solves the problem! I'm closing the issue.

closed 2023-04-17 by ABelliqueux