raspberrypi/rpicam-apps #503: Framerate is maxed at ~60fps using libcamera-vid, rpi cam module v3 and raspbian bullseye on rpi 3b

ABelliqueux, 2023-04-17:
## Describe the bug

When trying the high-framerate examples from the documentation (https://www.raspberrypi.com/documentation/computers/camera_software.html#high-framerate-capture), the framerate in the resulting file is not 120fps.

I have followed the various suggestions in the doc linked above ( force_turbo, gpu_freq, etc. ).

Although `dmesg` reports the correct timings between frames (~8.33ms) after setting '/sys/class/video4linux/video0/dev_debug' to 3, **the resulting h264 and pts files only show 575 frames ( 57.5fps vs 120fps ).**

## Steps to reproduce the issue

1. Use the recording command from the doc:
```
libcamera-vid --level 4.2 --framerate 120 --width 1280 --height 720 --save-pts timestamp.pts -o video.264 -t 10000 --denoise cdn_off -n
```
2. Check `dmesg` output (timing looks ok at ~8.33ms):  
```
[snip]
[ 2761.217657] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1196, id=0, timestamp=2761.206906811
[ 2761.217727] frame_sequence=1196
[ 2761.226098] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1197, id=0, timestamp=2761.215231811
[ 2761.226160] frame_sequence=1197
[ 2761.234930] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1198, id=0, timestamp=2761.223555613
[ 2761.234996] frame_sequence=1198
[ 2761.242531] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1199, id=0, timestamp=2761.231880925
[ 2761.242586] frame_sequence=1199
[ 2761.251044] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1200, id=0, timestamp=2761.240212331
[ 2761.251121] frame_sequence=1200
[ 2761.259298] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1201, id=0, timestamp=2761.248530248
[ 2761.259350] frame_sequence=1201
[ 2761.267522] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1202, id=0, timestamp=2761.256855300
[ 2761.267591] frame_sequence=1202
[ 2761.275837] video0: VIDIOC_DQEVENT: type=0x4, pending=0, sequence=1203, id=0, timestamp=2761.265180925
[ 2761.275899] frame_sequence=1203
[ 2761.281838] video0: VIDIOC_UNSUBSCRIBE_EVENT: type=0x4, id=0x0, flags=0x0
[ 2761.283058] video0: VIDIOC_STREAMOFF: type=vid-cap
[ 2761.501125] video0: VIDIOC_REQBUFS: count=0, type=vid-cap, memory=dmabuf
[snip]
```
3. Check pts file length : 
```
$ wc -l /dev/shm/timestamp.pts
576 /dev/shm/timestamp.pts
```
4. Use doc's `mkvmerge` example to merge to mkv file:
```
mkvmerge -o test.mkv --timecodes 0:/dev/shm/timestamp.pts /dev/shm/video.264
```
5. Check mkv file length in frames ( NUMBER_OF_FRAMES: 575 ) : 
```
$ ffprobe test.mkv

[snip]

Input #0, matroska,webm, from 'test.mkv':
  Metadata:
    encoder         : libebml v1.4.4 + libmatroska v1.7.1
    creation_time   : 2023-04-17T10:36:18.000000Z
  Duration: 00:00:09.83, start: 0.000000, bitrate: 4880 kb/s
  Stream #0:0: Video: h264 (High), yuv420p(progressive), 1280x720, SAR 1:1 DAR 16:9, 60.06 fps, 60.06 tbr, 1k tbn (default)
    Metadata:
      BPS             : 4870581
      DURATION        : 00:00:09.832000000
      NUMBER_OF_FRAMES: 575
      NUMBER_OF_BYTES : 5985945
      _STATISTICS_WRITING_APP: mkvmerge v75.0.0 ('Goliath') 64-bit
      _STATISTICS_WRITING_DATE_UTC: 2023-04-17 10:36:18
      _STATISTICS_TAGS: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES
```

## Questions

  * Is my assumption that a 10 seconds 120 fps file should result in h264 and pts files containing 1200 frames / timestamps wrong ?
  * Is that speed not achievable on a RPI 3B ?

## Versions : 

Pi : `Pi Model 3B  V1.2, rev. a02082, Soc BCM2837, 1GB ram`

Camera : `rpi cam v3 ( imx708 [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a) )`

Kernel : `Linux raspberry 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`

libcamera apps : ```$libcamera-apps build: 54a781dffdd1 01-03-2023 (17:47:29)
libcamera build: v0.0.4+22-923f5d70```

OS: `Raspbian GNU/Linux 11`

Thanks !

naushir, 2023-04-18:
There is a timestamp analysis script here: https://github.com/raspberrypi/libcamera-apps/blob/main/utils/timestamp.py
Could you run it with
```
timestamp.py timestamp.pts
```
and report the output please?  As a test, you might also want to try sending the encoded stream output to ``/dev/null`` just to remove any effects of slow sdcard access.

6by9, 2023-04-18:
The H264 encoder spec is level 4.0, so 1080p30 or 720p60. This is what you'll find listed in all the product literature.
Under ideal conditions it can achieve level 4.2 (1080p60 or 720p120), but that is not guaranteed, and may require overclocking.

ABelliqueux, 2023-04-18:
Thanks for your replies !

@naushir:  

Following your instructions, I ran:  
`libcamera-vid --level 4.2 --framerate 120 --width 1280 --height 720 --save-pts /dev/shm/timestamp.pts -o /dev/null -t 10000 --denoise cdn_off -n` 
then:  
`timestamp.py /dev/shm/timestamp.pts`
which resulted in:
```
Total: 590 frames (589 samples)
Average: 16.680 ms / 59.952 fps
Minimum: 8.313 ms at frame 99
Maximum: 33.304 ms at frame 305
Outliers: 0 (100%) 195 (10.0%) 195 (1.0%) 588 (0.1%)
```

which shows mostly 16.6ms frames, not good :/  

@6by9 : This would mean the rpi 3b lacks the power needed to achieve this framerate ? 

Just out of curiosity, I ran the same command as above requesting a **640x480@90fps** video with 
`libcamera-vid --level 4.2 --framerate 90 --width 640 --height 480 --save-pts /dev/shm/timestamp.pts -o /dev/null -t 10000 --denoise cdn_off -n`

the rationale being that since I was able to record 640x480@90fps with the **old camera stack** (raspivid) and a v1 camera module, I should be able to obtain the same with libcamera, but the timestamp analysis show that even that level can't be achieved with libcamera:  

```
Total: 680 frames (679 samples)
Average: 14.442 ms / 69.244 fps
Minimum: 11.088 ms at frame 13
Maximum: 33.317 ms at frame 665
Outliers: 2 (100%) 679 (10.0%) 679 (1.0%) 679 (0.1%)
```

That's ~70fps, not good either :/  

### config.txt 

For full disclosure, here are the options I set in '/boot/config.txt' regarding overclocking : 

```
gpu_freq=504
over_voltage=6
# SD overclock to 84Mhz for more speed
# NEEDS A UHS-1/Class10 SD CARD !
dtparam=sd_overclock=84
# Disable bluetooth
dtoverlay=disable-bt
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2
force_turbo=1
```

The sdcard I'm using advertises itself as Class10 (oem:0x5048).  

I'm going to try to increase arm_freq and sdram_freq a bit to see if any difference occurs.   

Thanks to both of you for taking the time to look into this !

ABelliqueux, 2023-04-18:
After trying with a Pi 4, I can finally reach 120fps with the command above:  

```
Total: 1191 frames (1190 samples)
Average: 8.325 ms / 120.119 fps
Minimum: 8.314 ms at frame 1173
Maximum: 8.336 ms at frame 1172
Outliers: 0 (100%) 0 (10.0%) 0 (1.0%) 2 (0.1%)
```
so it looks like the Pi3 hardware is the culprit here, when used with libcamera, which I guess is not really fixable ?

naushir, 2023-04-18:
> so it looks like the Pi3 hardware is the culprit here, when used with libcamera, which I guess is not really fixable ?

Unfortunately not, this is a hardware limitation so software based workarounds are only going to help up to a point.

closed 2023-04-18 by ABelliqueux