raspberrypi/rpicam-apps #508: [BUG] Problem with high resolution

xXPerditorXx, 2023-05-26:
**Describe the bug**
If I want to go above the resolution of 1920x1080, libcamera-vid tells me: `ERROR: *** failed to start output streaming ***`

**Bug report**
`camera-bug-report -o bug.txt -c "libcamera-vid -t 0 --width 3840 --height 2160 --autofocus-mode manual -k --inline --nopreview -o udp://192.168.168.40:36830"` IP is my pc

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/11579775/bug.txt)

**Additional context**
Camera = Arducam 64MP Hawkeye
Raspberry Pi 4 8GB

naushir, 2023-05-26:
As you are using arducam hardware and their accompanying libraries that we have no source access to, please raise this on their support channels.

closed 2023-05-26 by naushir

6by9, 2023-05-27:
The H264 encoder is limited to level 4.0, or 1080p30. Level 4.1 (1080p60) can generally be achieved with overclocking.
There is a hard limit on the width at 2032 pixels.
https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels