raspberrypi/rpicam-apps #509: When the encoder of v4l2 uses the cbr mode to encode H264, the frame rate is low[BUG]

q5270114, 2023-05-27:
each frame consumes approximately 100ms.

respberry pi 4b.
os: 
[image]

this is my encode code :
//
// Created by admin on 2022/11/22.
//
#ifdef ENABLE_V4L2_HARDWARE_ENCODE
#include "V4l2Encoder.h"
#include "spdlog/spdlog.h"
int xioctl(int fd, unsigned long ctl, void *arg)
{
    int ret, num_tries = 10;
    do
    {
        ret = ioctl(fd, ctl, arg);
    } while (ret == -1 && errno == EINTR && num_tries-- > 0);
    return ret;
}

V4l2Encoder::V4l2Encoder(int width, int height, long bitrate, int fps, Callback cb) {
    this -> cb = cb;
    const char device_name[] = "/dev/video11";
    fd_ = open(device_name, O_RDWR, 0);
    if (fd_ < 0)
        throw std::runtime_error("failed to open V4L2 H264 encoder");
    spdlog::info("Opened V4L2 H264Encoder on {}  as fd {}", device_name, fd_);

    v4l2_control ctrl = {};
    // set bitrate mode
    ctrl.id = V4L2_CID_MPEG_VIDEO_BITRATE_MODE;
    ctrl.value = V4L2_MPEG_VIDEO_BITRATE_MODE_CBR;
    if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
        throw std::runtime_error("failed to set bitrate");

    // set bitrate
    if (bitrate) {
        ctrl.id = V4L2_CID_MPEG_VIDEO_BITRATE;
        ctrl.value = bitrate;
        if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
            throw std::runtime_error("failed to set bitrate");
    }

    // set profile
    ctrl.id = V4L2_CID_MPEG_VIDEO_H264_PROFILE;
    ctrl.value = V4L2_MPEG_VIDEO_H264_PROFILE_BASELINE;
    if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
        throw std::runtime_error("failed to set profile");

    // set level
    ctrl.id = V4L2_CID_MPEG_VIDEO_H264_LEVEL;
    ctrl.value = V4L2_MPEG_VIDEO_H264_LEVEL_4_2;
    if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
        throw std::runtime_error("failed to set level");

    // set i frame period
    ctrl.id = V4L2_CID_MPEG_VIDEO_H264_I_PERIOD;
    ctrl.value = fps;
    if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
        throw std::runtime_error("failed to set intra period");

    // set seq header
    ctrl.id = V4L2_CID_MPEG_VIDEO_REPEAT_SEQ_HEADER;
    ctrl.value = 1;
    if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
        throw std::runtime_error("failed to set inline headers");

[... 265 lines cut ...]
                OutputItem item = { buffers_[buf.index].mem,
                                    buf.m.planes[0].bytesused,
                                    buf.m.planes[0].length,
                                    buf.index,
                                    !!(buf.flags & V4L2_BUF_FLAG_KEYFRAME),
                                    timestamp_us };
                std::lock_guard<std::mutex> lock(output_mutex_);
                output_queue_.push(item);
                output_cond_var_.notify_one();
            }
        }
    }
}

void V4l2Encoder::setBitrate(long bitrate) {
    v4l2_control ctrl = {};
    // set bitrate
    if (bitrate) {
        ctrl.id = V4L2_CID_MPEG_VIDEO_BITRATE;
        ctrl.value = bitrate;
        if (xioctl(fd_, VIDIOC_S_CTRL, &ctrl) < 0)
            throw std::runtime_error("failed to set bitrate");
    }
}
#endif//ENABLE_V4L2_HARDWARE_ENCODE

6by9, 2023-05-28:
Your quoted code is insufficient for anyone to reproduce your test case. You've given no information as to what resolution you're asking to encode for a start.

The 2 queues should normally be treated independently, not with DQBUF being called on both based on the one event. Due to pipelining and internal buffering the input frame is finished with quite a while before the encoded frame is ready. At 1080p, memory says that the full frame encode takes around 45ms.

6by9, 2023-06-01:
> Do you mean that 1080p requires 45ms per frame in CBR mode? So, in CBR mode, the encoder can only encode 1080P 22FPS, right?

No, it is pipelined.
The hardware need images in a particular format, so the first stage is an image conversion.
The second stage is generating motion estimation values for all potential options.
The third is the choice of motion vector
The fourth is entropy coding (CAVLC or CABAC)

Memory says that steps 2&3 are done as one job, but that still leaves a potential pipeline of 3 frames. Max frame rate is the time taken for the longest of the steps, not the total.
What does fail badly with a pipelined codec is feeding one frame in and waiting for the output frame before feeding the next one in.

You've created this as an issue under libcamera-apps. Are you actually using libcamera-apps, or just the V4L2 encoder?
If libcamera-apps, why the need to create your own wrapper around the H264 encoder?

q5270114, 2023-06-02:
> 

Thanks for your reply, we have a requirement to use CBR mode hard encoder encoded h264 (1080P 30FPS) live broadcast, and we need to adjust the bit rate in real time. Since there are few hard coding data about raspberry pie, I can only choose to modify the examples in libcamera apps to support our own business implementation. Do you have any good suggestions for me to modify my test cases to achieve this goal? By the way, when I use the ffmpeg command line tool to code, the hard encoder's encoding speed is relatively fast, and the bit rate control is very stable

q5270114, 2023-06-12:
> No, it is pipelined. The hardware need images in a particular format, so the first stage is an image conversion. The second stage is generating motion estimation values for all potential options. [...]

closed 2023-06-12 by q5270114