raspberrypi/rpicam-apps #510: [HELP] Post-processing example for OpenCV and/or TensorFlow Lit

baqwas, 2023-06-01:
**Describe the issue**
Cannot find any code snippet related to post-processing with OpenCV and/or TensorFlow Lite at [https://www.raspberrypi.com/documentation/computers/camera_software.html#getting-help](https://www.raspberrypi.com/documentation/computers/camera_software.html#getting-help)

**Help request**

- I have followed all the steps successfully to prepare for post-processing with OpenCV and TFL.
- I have _built_ **libcamera** as outlined in the documentation
- I have _built_ (from source v4.5.3) OpenCV
- I have confirmed simple image processing works with the OpenCV deployment including primitive object detection
- I have built TFL apps for object detection and classification for Android devices (that use OpenCV for image capture and Google online models)

Is there a very simple code snippet that illustrates post-processing of libcamera output using OpenCV and/or TFL? I understand that it is impractical to accommodate diverse examples but something very simple to ensure a practical pipeline would be very helpful for my continuing self-paced exercises.

My predicament is that several JSON files are mentioned as being used during the **libcamera** image acquisition process. Unfortunately, I have no idea how to leverage these files when running OpenCV. If the libcamera team has done some work on this topic would it be possible to share a few lines of skeleton code snippet to get a _newbie_ started? Thanks.

Kind regards.

naushir, 2023-06-02:
I presume you have looked at the relevant documentation sections, particularly [this](https://www.raspberrypi.com/documentation/computers/camera_software.html#post-processing-with-opencv) and [this](https://www.raspberrypi.com/documentation/computers/camera_software.html#post-processing-with-tensorflow-lite)?

All our example postprocessing stages that use either TFlite or OpenCV can be found in the [post_processing_stages](https://github.com/raspberrypi/libcamera-apps/tree/main/post_processing_stages) subdirectory of this repo.

Depending on what you are doing, it might also be worth looking at [picamera2](https://github.com/raspberrypi/picamera2).  There are some examples of OpenCV postprocessing that are available in the repo as well.  You can find a comprehensive manual on picamera2 [here](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

baqwas, 2023-06-02:
Thanks @naushir for the heads up. I do have the links you provided. I simply cannot tie the loose ends (_my issue_). I'll step through the documentation again step by step.

I understand the advantages of integrating libcamera for OpenCV and TFL use. My earlier use of pure OpenCV for object detection was _laggy_. I will definitely keep trying. Thanks again for the excellent work on **libcamera-apps**.

Regards.

closed 2023-06-02 by baqwas